sipeed/NanoKVM #266: Web interface is gone (again)

aka-charos, 2025-01-18:
web interface is gone. When I try to restart it I get 

`# /etc/init.d/S95nanokvm restart
killall: NanoKVM-Server: no process killed
OK
# killall: NanoKVM-Server: no process killed
rm: can't remove '/tmp/server': No such file or directory
beta
oled exist
192.168.178.84
rm: can't remove '/etc/kvm/wifi_exist': No such file or directory
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
try release sys ok
[_mmf_init]-1142: maix multi-media version:67a90e2-musl_riscv64
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
pstSize->u32Width = 57462
pstSize->u32Height = 3043
IOCTL_VB_CMD - vb_ioctl_init NG
[_mmf_sys_init]-930: system init failed with 0xffffffff
[_mmf_init]-1194: sys init failed. s32Ret: 0xffffffff !
maix multi-media init failed`

I am not sure how/why incorrect video buffer initialization is preventing the interface to load.

bkmartin, 2025-01-20:
i see the same thing

/etc/init.d/S95nanokvm restart
killall: NanoKVM-Server: no process killed
OK
# killall: NanoKVM-Server: no process killed
rm: can't remove '/tmp/server': No such file or directory
PCIe
oled exist
192.168.1.92
192.168.1.215
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
try release sys ok
[_mmf_init]-1142: maix multi-media version:67a90e2-musl_riscv64
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
pstSize->u32Width = 0
pstSize->u32Height = 1
IOCTL_VB_CMD - vb_ioctl_set_config NG
[_mmf_sys_init]-930: system init failed with 0xffffffff
[_mmf_init]-1194: sys init failed. s32Ret: 0xffffffff !
maix multi-media init failed

bkmartin, 2025-01-20:
Jan 20 16:40:40 kvm-9741 kern.info kernel: [   22.525958] NanoKVM-Server[627]: unhandled signal 11 code 0x1 at 0x00000000000001a0 in libstdc++.so.6.0.28[3fd28a0000+183000]
Jan 20 16:40:40 kvm-9741 kern.warn kernel: [   22.526007] CPU: 0 PID: 627 Comm: NanoKVM-Server Tainted: GF          O      5.10.4-tag- #31
Jan 20 16:40:40 kvm-9741 kern.warn kernel: [   22.526017] epc: 0000003fd29a98bc ra : 0000003fd4aa9c08 sp : 0000003ffff937b0
Jan 20 16:40:40 kvm-9741 kern.warn kernel: [   22.526025]  gp : 0000000000d4b858 tp : 0000003fd28510e8 t0 : 0000000000000039
Jan 20 16:40:40 kvm-9741 kern.warn kernel: [   22.526033]  t1 : 0000003fd4a9e08c t2 : 0000003fd4c6a390 s0 : 0000003ffff937d0
Jan 20 16:40:40 kvm-9741 kern.warn kernel: [   22.526064]  s1 : 0000003ffff937d0 a0 : 0000003ffff937d0 a1 : 00000000000001a0
Jan 20 16:40:40 kvm-9741 kern.warn kernel: [   22.526073]  a2 : 0000000000000001 a3 : 0000003ffff937e0 a4 : 0000003ffff93870
Jan 20 16:40:40 kvm-9741 kern.warn kernel: [   22.526081]  a5 : 00000000000001a0 a6 : 000000006d2d6974 a7 : 0000000000000042
Jan 20 16:40:40 kvm-9741 kern.warn kernel: [   22.526089]  s2 : 0000003fd2c3f1f8 s3 : 0000003ffff93860 s4 : 0000000000000140
Jan 20 16:40:40 kvm-9741 kern.warn kernel: [   22.526097]  s5 : 0000000000000000 s6 : 0000000000000001 s7 : 0000000000000003
Jan 20 16:40:40 kvm-9741 kern.warn kernel: [   22.526106]  s8 : 0000000000000001 s9 : 0000003ffff93a40 s10: 0000003ffff93a20
Jan 20 16:40:40 kvm-9741 kern.warn kernel: [   22.526114]  s11: 0000003ffff93a00 t3 : 0000003fd29a98b4 t4 : 0000000000000010
Jan 20 16:40:40 kvm-9741 kern.warn kernel: [   22.526121]  t5 : 0000000073690000 t6 : 0000000049560000
Jan 20 16:40:40 kvm-9741 kern.warn kernel: [   22.526129] status: 8000000201804020 badaddr: 00000000000001a0 cause: 000000000000000d

chchia, 2025-01-22:
and this is the 4th time for me, CUBE full

```
# /etc/init.d/S95nanokvm restart
killall: NanoKVM-Server: no process killed
OK
# Failed to connect to local Tailscale daemon for /localapi/v0/status; not running? Error: dial unix /var/run/tailscale/tailscaled.sock: connect: no such file or directory
killall: NanoKVM-Server: no process killed
rm: can't remove '/tmp/server': No such file or directory
beta
oled exist
192.168.1.4
rm: can't remove '/etc/kvm/wifi_exist': No such file or directory
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
try release sys ok
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^reinit_soph_vb
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^reinit_soph_vb
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^reinit_soph_vb
[_mmf_init]-1142: maix multi-media version:67a90e2-musl_riscv64
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
pstSize->u32Width = 16
pstSize->u32Height = 1
alloc_buf_size(294912): 294912, phy_addr(8b301000): 0x8b301000, buffer_name: ISP_SHARED_BUFFER_0
ion_data.size:294912, ion_data.addr_p:0x8b301000, ion_data.name:ISP_SHARED_BUFFER_0
pstSize->u32Width = 16
pstSize->u32Height = 1
pstSize->u32Width = 16
pstSize->u32Height = 1
ISP Vipipe(0) Allocate pa(0x8b301000) va(0x0x3ff28c0000) size(291424)
pstSize->u32Width = 16
pstSize->u32Height = 1
stSnsrMode.u16Width 16 stSnsrMode.u16Height 1 60.000000 wdrMode 0 pstSnsObj 0x3ff2e9b2e8
pstSize->u32Width = 16
pstSize->u32Height = 1
pstSize->u32Width = 16
pstSize->u32Height = 1
[SAMPLE_COMM_VI_StartMIPI]-494: sensor 0 stDevAttr.devno 0
[sensor_prob = lt6911_probe]
###########   PinMux  #######################################################################
###########   PinMux End  #######################################################################
[lt6911_probe] -189
[lt6911_probe] -195
[lt6911_i2c_write] : addr = ee. data = 1
[lt6911_i2c_read] : addr = 0
[lt6911_i2c_write] : addr = ee. data = 0
[lt6911_i2c_write] : addr = ee. data = 1
[lt6911_i2c_read] : addr = 1
[lt6911_i2c_write] : addr = ee. data = 0
[lt6911_probe] jump return nVal -198
[lt6911_probe] -201
data:ffffffff ffffffff
[lt6911_probe] jump return CVI_FAILURE -206
[lt6911_probe] -208 = exit success
pstSize->u32Width = 16
pstSize->u32Height = 1
20250122 14:57:57.525 1540 E isp AE_InitSensorFpsPara:5555 pfn_cmos_fps_set NULL!

AE_SetExposure inttime update NULL.
awbInit ver 6.9@2021500
0 R:1400 B:3100 CT:2850
1 R:1500 B:2500 CT:3900
2 R:2300 B:1600 CT:6500
Golden 1024 1024 1024
WB Quadratic:0
isWdr:0
20250122 14:57:57.534 1540 E isp setNoiseProfile:3460 Noise profile get fail. Please check
********************************************************************************
cvi_bin_isp message
gerritId:      97347          commitId:      fc6ce647
md5:           d6db2297ddfd44e8252c1f3f888f47b2
sensorNum      1
sensorName0    6911

PQBIN message
gerritId:
              commitId:      fc6ce647
md5:           d6db2297ddfd44e8252c1f3f888f47b2
sensorNum      1
sensorName0    4653

author:        lxowalle       desc:          gc4653 30fps
createTime:    2024-03-26 13:41:34version:
tool Version:       v3.0.5.48           mode:
********************************************************************************
sensorName(0) mismatch, mwSns:6911 != pqBinSns:4653
[SAMPLE_COMM_ISP_Thread]-95: ISP Dev 0 running!
pstSize->u32Width = 16
pstSize->u32Height = 1
maix multi-media init ok
try release vio ok
try release venc ok
[_mmf_vpss_init_new]-1490: CVI_VPSS_CreateGrp(grp:0) retry(0xc0068003)!
[_mmf_vpss_init_new]-1495: CVI_VPSS_CreateGrp(grp:0) failed with 0xc0068003!
[mmf_vi_init]-1523: _mmf_vpss_init_new failed. s32Ret: 0xffffffff !

```

vonasek, 2025-01-23:
I ran into a similar issue with my NanoKVM, where the NanoKVM-Server kept crashing during startup due to an invalid resolution (26x11). I managed to fix it by adjusting the resolution values in /kvmapp/kvm like this:

```
echo 1920 > /kvmapp/kvm/width
echo 1080 > /kvmapp/kvm/height
/etc/init.d/S95nanokvm restart
```
After that, the NanoKVM-Server started up without any problems. Hopefully, this helps someone facing the same issue!

hardingt, 2025-02-03:
@vonasek That fixed it for me as well. Saved me a lot of time getting at the SD Card.

Z2Z-GuGu, 2025-03-04:
I found that in certain specific situations, HDMI seems to not receive the correct resolution information, which interferes with the startup of the VI. In a recent update, I attempted to fix the issue. Please update to the latest app, which is currently version 2.2.1. Then, execute the following command in the web terminal: echo 1 > hdmi_mode && reboot. After that, wait for it to boot up, and it should display the image again shortly.

bkmartin, 2025-03-04:
from looking at the code, the hdmi_mode file needs to be in /etc/kvm

the instructions above were incomplete on that