sipeed/NanoKVM #269: Set static IP, now can't access

dube920, 2025-01-19:
NanoKVM PCI was working fine on 192.168.1.180 via DHCP. Set static IP to 192.168.1.240 according to docs (https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/network/static_ip.html) and it seemed to have worked- OLED is displaying new IP, router shows the NanoKVM online at new IP, arp-a from other devices shows the new IP.

But pinging, SSH, and accessing the WebUI on the new IP now fails. Same with the old IP. Seems the unit is just inaccessible or even bricked now? Tried a hard reboot of my router and the server where the NanoKVM is installed, no change. Help!

yamrod, 2025-01-28:
What did you set as the contents of the eth.nodhcp file?

ivahos, 2025-09-01:
I had the same issue. adding the /boot/eth.nodhcp set the ip but not the default gw.
Turned out that the S30eth init.d script does not handle the comments the example uses.
Make sure that there is no comment in the file. Just ipaddr/netmask routperip

dormancygrace, 2026-09-19:
Hi! [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS/releases/tag/v2.0-a2), my independent community firmware for SG2002-based NanoKVM devices, includes a change relevant to this request.

The Ethernet startup script and Web API parser now handle comments and empty lines in eth.nodhcp. Persistent static IPv4 configuration is also available in the Web UI.

[Implementation 1](https://github.com/dormancygrace/NanoKVM-OS/blob/bf7d726f502526abb127f687ca3a868487f39b25/kvmapp/system/init.d/S30eth) · [Implementation 2](https://github.com/dormancygrace/NanoKVM-OS/blob/bf7d726f502526abb127f687ca3a868487f39b25/server/service/network/ethernet.go).

You're welcome to try **NanoKVM OS v2.0-a2** and let me know how it works on your setup!

[Release and installation details](https://github.com/dormancygrace/NanoKVM-OS/releases/tag/v2.0-a2)

Please include your board model, OS version and reproduction steps when reporting results. Thanks!