sipeed/NanoKVM #270: Some security issues

MatejKovacic, 2025-01-20:
This device is really great, however there is a lot of work for security improvement. I completely understand that this is not a priority right now - first are usability and performance issues - but I hope developers will address these issues soon.

Here are some problems (this is not a complete list, just a quick observation).

1. If no account configuration file is found, it defaults to a predefined admin account (admin/admin). Better solution would be, that when you run the NanoKVM the first time, you would need to set up it - and set strong password.

2. By default `ssh` is enabled, and has default `root` password (`root`). Better solution would be to enable `ssh` via web interface and set initial password there.

3. `EncryptSecretKey` is hardcoded for all "encrypt" operations.

4. There is no CSRF protection on websocket and API endpoints.

5. Auth token does not have refresh call, TTL is 30 days.

6. User can not invalidate session (if session gets stolen).

7. There is a lot of cases where developers decided to use sleep function between operations. The problem is, that it is possible that there could be situations, where the previous operation does not end in the sleep time. This can lead to weird and unpredictable situations.

8. Why is NanoKVM sending your device key to the Sipeed?

9. Network problems. Some users are using their own Wireguard servers (see [issue here](https://github.com/sipeed/NanoKVM/issueswire)). This works, however I found out that the Wireguard version on NanoKVM does not work in some networks with more restricted frirewalls. In my company they have some restrictions in place (at least I know that they are blocking 3rd party DNS servers), and NanoKVM does not work there. I mean, it works, but only locally, and can not connect to the internet (i. e. can  not check for updates), and Wireguard is also not working. On the other hand, Wireguard version on my computer works fine.

10. NanoKVM is coneccting (and presumalby downloading something) from here: `https://wiki.sipeed.com/maixvision`. Why?

11. NanoKVM has `tcpdump` and `aircrack` installed. (Seee [issue here](https://github.com/sipeed/NanoKVM/issues/248). I understand that is needed for development, but for production version sounds bad.

12. Download code (for updates) does not check the integrity, it just downloads zip file.

13. API is running as root. My suggestion would be to isolate this as well...

14. Device uses some "custom" DNS servers and it is quite hard for users to change that... There should be network settings in the web GUI with all the transparent information. Also, option to use proxy would be nice.

This is not an exhaustive list, just some random observations, which show, that there needs to be done a lot in the area of security.

Also, I would suggest that instead of current Linux version you use [this port of Debian](https://github.com/Fishwaldo/sophgo-sg200x-debian/). It is using `apt` (so you can normally install software - for instance right now there is no OpenVPN package on the device), `systemd` (so you can run services normally), etc.

My suggestion would be to help this developer with porting Debian to NanoKVM board (basically it is just `SG200x board`) and cooperate with Debian so that `SG200x boards` would be officially supported by Debian. That will increase trust in devices, make them much more usable and also lower the cost of development (because underlying Linux would be supported by community).
 
Also, it would be nice to have some [security review](https://www.scmp.com/news/china/science/article/3265493/scientists-find-security-risk-risc-v-open-source-chip-architecture-china-hopes-can-help-sidestep-us) of RISC-V CPU (like [this one](https://www.researchgate.net/publication/353086020_A_Survey_on_RISC-V_Security_Hardware_and_Architecture)), but OK, I understand this is a little out of the scope at this point.

MatejKovacic, 2025-01-20:
Ah, and also... I would suggest to have an optin for modules. In GUI. That would allow users to install module for OpenVPN, custom Wireguard settings, etc.

devoxel, 2025-01-20:
> NanoKVM is coneccting (and presumalby downloading something) from here: https://wiki.sipeed.com/maixvision. Why?

Reading the source, it appears to be [downloading](https://github.com/sipeed/NanoKVM/blob/main/server/service/application/application.go#L201) a shared binary object - `libmaixcam_lib.so` [[^](https://github.com/sipeed/NanoKVM/blob/1db304a5f36e07974ee93bce2758779d5f66fdc5/server/service/application/service.go#L11)] which it then detects and places into [/kvmapp/kvm_system/dl_lib](https://github.com/sipeed/NanoKVM/blob/1db304a5f36e07974ee93bce2758779d5f66fdc5/support/kvm_system/main/lib/system_init/system_init.cpp#L131).

The purpose of this library is referenced [here](https://github.com/sipeed/NanoKVM/issues/1#issuecomment-2433424418). It is not clear why this library is downloaded on boot instead of being included.

In any case, **it's very insecure** - since if an attacker gains control of the endpoint they can place malware in the shared binary. These binary sources should be at least be checked to ensure they match a specified hash sum; but ideally, these binaries could just be included in the release.

scpcom, 2025-01-20:
>     1. If no account configuration file is found, it defaults to a predefined admin account (admin/admin). Better solution would be, that when you run the NanoKVM the first time, you would need to set up it - and set strong password.

This is already done in the latest version. If you login to the web interface you get at notification to set the password. If you change the password the ssh password will also be changed.

>     8. Why is NanoKVM sending your device key to the Sipeed?

This is required by the server to generate the downloaded libmaixcam_lib.so which is locked to work only on this device. Which is also the reason why it is not included in the image.

> The purpose of this library is referenced [here](https://github.com/sipeed/NanoKVM/issues/1#issuecomment-2433424418). It is not clear why this library is downloaded on boot instead of being included.

If you follow the link which [devoxel](https://github.com/devoxel) already mentioned, you find a fully functional open source replacement. I do not know why Sipeed still prefers to ship a dongled version, there is no secret inside anymore :-)

MatejKovacic, 2025-01-20:
So, if I understand this `libmaixcam_lib.so`, which is locked for a device could be replaced with this `streameye/kvm_stream`?

Well, I would be really happy if someone would make a truly opensource version of this, based on [this Debian port](https://github.com/Fishwaldo/sophgo-sg200x-debian/) and code mentioned here.

rtfmkiesel, 2025-01-21:
I investigated the download of `libmaixcam_lib.so` a bit yesterday. It was a suspicious to me that the endpoint was called `encryption`. (see [this line](https://github.com/sipeed/NanoKVM/blob/c0d9d167a78b50d8316eca25b43e76e3413ba5dd/server/service/application/service.go#L6))

This is also the endpoint that receives you `/device_key` as a `uid` GET parameter.

If you want to test this, use the following command: (Linux, add `.exe` to `curl` for Windows)

```sh
curl "https://maixvision.sipeed.com/api/v1/nanokvm/encryption?uid=<YOUR DEVICE KEY>" -H "token: MaixVision2024" -o libmaxicam_lib.so
```

Depending on your device key, this file differs. This means, each user will get a unique shared object. I verified this using checksums. This is very weird. 

If you want to verify this, here is a list of device keys, which I guessed by comparing two device keys of NanoKVM's:

```text
d1c94d9c1359c262 
d1c94d9e1359c262 
d1c94da21359c262 
d1c94da41359c262 
d1c94da61359c262 
d1c94daa1359c262 
d1c94dac1359c262 
d1c94dae1359c262 
d1c94db01359c262 
d1c94db21359c262 
d1c94dba1359c262 
d1c94dbc1359c262 
d1c94e031359c262 
d1c94e041359c262 
d1c94e061359c262 
d1c94e0a1359c262 
d1c94e0c1359c262 
d1c94e0e1359c262 
d1c94e131359c262 
d1c94e171359c262 
d1c94e191359c262 
d1c94e1b1359c262 
d1c94e1d1359c262 
d1c94e1f1359c262 
d1c94e231359c262 
d1c94e251359c262 
d1c94e271359c262 
d1c94e291359c262 
d1c94e2e1359c262 
d1c94e2f1359c262 
d1c94e311359c262 
d1c94e331359c262 
d1c94e371359c262 
```

I'm not into reverse engineering (RISC-V) Linux stuff. Maybe someone wants to investigate this further. The ELF header with `readelf -a` seems to be the same. It might also just be a freshly compiled shared object each time someone requests this API endpoint, hence the checksum difference.

wj-xiao, 2025-01-22:
Thank you for raising these questions. Enhancing security will be our primary focus in the upcoming development phases.

The issue with libmaixcam_lib.so is a legacy matter. We reused code from our previous project (MaixCAM), which is why this dependencie is still present.
Currently, some HDMI features rely on this library. We are planning to refactor this part of the code and aim to remove the dependency by Q1 2025.

wj-xiao, 2025-01-22:
A brief response to the above questions：

> 1. If no account configuration file is found, it defaults to a predefined admin account (admin/admin). Better solution would be, that when you run the NanoKVM the first time, you would need to set [...]

Now we prompt users to change their passwords. And the changed password will apply to both web and SSH access. We are also considering disabling SSH access by default.

> 3. EncryptSecretKey is hardcoded for all "encrypt" operations.

This `EncryptSecretKey` is only used to encrypt password strings, preventing plaintext transmission of passwords.

> 4. There is no CSRF protection on websocket and API endpoints.

Enabling CSRF may cause api unavailability in some cases. We will fix this issue.

> 5. Auth token does not have refresh call, TTL is 30 days.
> 6. User can not invalidate session (if session gets stolen).

The JWT token key is dynamically generated upon each service restart, therefore, all existing logins will be invalidated.
You can also configure a static key in the configuration file to avoid this behavior.

> 7. There is a lot of cases where developers decided to use sleep function between operations. The problem is, that it is possible that there could be situations, where the previous operation does not end in the sleep time. This can lead to weird and unpredictable situations.

This is bug and we will fix it.

> 8. Why is NanoKVM sending your device key to the Sipeed?

To download libmaixcam_lib.so, please refer to my previous comment.

> 9. Network problems. Some users are using their own Wireguard servers (see [issue here](https://github.com/sipeed/NanoKVM/issueswire)). This works, however I found out that the Wireguard version on NanoKVM does not work in some networks with more restricted frirewalls.n my company they have some restrictions in place (at least I know that they are blocking 3rd party DNS servers), and NanoKVM does not work there. I mean, it works, but only locally, and can not connect to the internet (i. e.
can not check for updates), and Wireguard is also not working. On the other hand, Wireguard version on my computer works fine.

we will fix it.

> 10. NanoKVM is coneccting (and presumalby downloading something) from here: https://wiki.sipeed.com/maixvision. Why?

To download libmaixcam_lib.so.

> 11. NanoKVM has tcpdump and aircrack installed. (Seee https://github.com/sipeed/NanoKVM/issues/248. I understand that is needed for development, but for production version sounds bad.
> 12. Download code (for updates) does not check the integrity, it just downloads zip file.
> 13. API is running as root. My suggestion would be to isolate this as well...
> 14. Device uses some "custom" DNS servers and it is quite hard for users to change that... There should be network settings in the web GUI with all the transparent information. Also, option to use proxy would be nice.

We will optimize these issues later.

MatejKovacic, 2025-01-22:
Great! I think this is a great product and I am glad it will be even better.

gshpychka, 2025-01-22:
> To download libmaixcam_lib.so, please refer to my previous comment.

Why is the device key required to download?

gshpychka, 2025-01-22:
> > Why is the device key required to download?
> 
> Device key verification is required before downloading. An incorrect key will prevent the download.

Right - I'm asking why?

ciroiriarte, 2025-01-24:
> > > Why is the device key required to download?
> > 
> > 
> > Device key verification is required before downloading. An incorrect key will prevent the download.
> 
> Right - I'm asking why?

Making sure it's only run on their hardware?, maybe to avoid support requests from users running unknown hardware?

gshpychka, 2025-01-27:
> > > > Why is the device key required to download?
> > > 
> > > 
> > > Device key verification is required before downloading. An incorrect key will prevent the download.
> > 
> > 
> > Right - I'm asking why?
> 
> Making sure it's only run on their hardware?, maybe to avoid support requests from users running unknown hardware?

Maybe! I'm sure @wj-xiao can provide the actual answer

rtfmkiesel, 2025-01-29:
> This EncryptSecretKey is only used to encrypt password strings, preventing plaintext transmission of passwords.

This is useless if the key is hard coded.

MatejKovacic, 2025-01-29:
@scpcom I tested your build and is working really nice (I have original Lichee RV Nano device). However, I have some questions regarding RNDIS and hostname), and you do not have issues enabled on your Github repo. How can I contact you?

scpcom, 2025-01-29:
The RNDIS addon is from the original project, you may look here for Issues:
https://github.com/Fishwaldo/sophgo-sg200x-debian/issues
The hostname generation on my fork is a script from Sipeed LicheeRV-Nano-Build, you can write your own hostname in to a new file named /boot/hostname
I may enable the issues tab on my fork later, at this moment this is just a test system.

MatejKovacic, 2025-01-30:
Thanks. Actually I like your fork, because user LED can be controlled (`/sys/class/leds/led-user/brightness`). Sorry for my "kidnapping" of this issue...

apalrd, 2025-02-04:
> I'm not into reverse engineering (RISC-V) Linux stuff. Maybe someone wants to investigate this further. The ELF header with `readelf -a` seems to be the same. It might also just be a freshly compiled shared object each time someone requests this API endpoint, hence the checksum difference.

I'm into reverse engineering, so I took a crack at it in Ghidra. 

There are 7 differences between binaries from different serial numbers. They are all exactly 64 bytes long, in a .text section, but contain apparently random data (does not disassemble into reasonable code). They are all immediately preceeded by an (indirect) call to auth_func, followed by an unconditional jump around the weird data. There are no labels anywhere which actually point to these 7 data blobs. Each of these 7 regions has different data. The functions they are used in are:
- maixcam_load_cvimodel
- maix::nn::NN_MaixCam::forward
- maix:nn::NN_MaixCam::forward_image
- mmf_add_vi_channel
- maix::nn:NN_MaixCam::load
- maix_nn_self_learn_classifier_learn
- An unknown function which is called from forward()

[image]

(pre_auth_func is a name that I gave to the stub which loads the auth_func pointer and jumps to it, but auth_func is the actual name in ELF). At this point, the data (in this example at 142540) could be referenced by the return address register (ra), which will point to 14253e, as it is incremented beyond the jump-and-link. So, in auth_func, we look for use of the ra register as data.

[image]

Within auth_func, the ra register is pushed/popped and eventually used as the return address (so we do not try to execute the new 64 byte of 'code'). It's also stored in s1, which is pushed/popped by get_uid, and then incremented by 2 (to skip over the unconditional jump) as a pointer to the data, and passed as an argument to get_uidl. Both get_uid and get_uidl are also indirect via a pointer table, hence the t_get_uid and t_get_uidl names I gave them. 

In get_uid, it calls open() on /dev/mem, followed by two calls to mmap(). There are two instructions (well, the same instruction, used twice) here that both Ghidra and Binutils are unable to disassemble, and I was unable to find any instruction using those opcodes on the internet either. It's either from a relatively new instruction extension or it's a custom instruction from Cvitek. get_uid decompiled well, so here's my annotated decompilation:

[image]

get_uidl does not decompile well due to apparently unreachable code. 

[image]

This time, the custom instruction is 0x8b 57 c7 00. This function is a simple for loop which iterates from 0 to 7 (8 times). Each time, it reads 8 unsigned bytes off the pointer (to weird_data), adds them, and then calls the custom intstruction. 

[image]

The data address is in a0, each loop it lbu (load byte unsigned)'s the 8 bytes, adds them all into register a5, increments the pointer by 8, then the custom instruction presumably does something special with the data now in a5. This also means that the 64-bytes of random data is actually only 8 chunks of random data with a range of 0-2047, since they are loaded as unsigned bytes and added as words. This gives us 88 bits of 'password' in the auth function, but consuming 512 bits (64 bytes) in the file.

Given that the bytes are just added, I tried adding them up and seeing if the 7 occurrences are actually equal. They are not, but their lowest bytes are equal from all 7 locations, so maybe they are only considering the lowest bytes.

`Chunk      locations
Chunk 0 is 362 462 462 362 462 362 362 
Chunk 1 is 2c2 3c2 3c2 3c2 2c2 2c2 4c2
Chunk 2 is 359 359 559 359 359 359 459
Chunk 3 is 313 413 513 413 413 313 413
Chunk 4 is 5a2 2a2 3a2 3a2 3a2 2a2 4a2
Chunk 5 is 44d 24d 34d 44d 44d 44d 24d
Chunk 6 is 4c9 4c9 3c9 4c9 5c9 2c9 4c9
Chunk 7 is 4d1 4d1 3d1 2d1 4d1 3d1 4d1`

So presumably this is some sort of licensing thing? I definitely do not feel comfortable with any of this in an 'open source' product.

Zepan, 2025-02-05:
> > I'm not into reverse engineering (RISC-V) Linux stuff. Maybe someone wants to investigate this further. The ELF header with `readelf -a` seems to be the same. It might also just be a freshly compiled shared object each time someone requests this API endpoint, hence the checksum difference.
> 
> So presumably this is some sort of licensing thing? I definitely do not feel comfortable with any of this in an 'open source' product.

Hi, NanoKVM project is a side project forked from our commercial project  "MaixCAM" (as you can see maixcam_load_cvimodel, etc)
For the sake of project progress, we have adopted the pre-packaged runtime libraries and their authentication mechanisms from the MaixCAM project.

Currently, the first generation of KVM has largely been finalized. We plan to decouple these runtime libraries from the original project before the launch of the second generation, NanoKVM Pro. In fact, developers within the community have already independently implemented the functionality of these interfaces: https://github.com/scpcom/LicheeSG-Nano-Build.

MatejKovacic, 2025-02-05:
Interesting. Just a couple of questions.
1. Will be software updated also for NanoKVM (not just Pro version)? Will we receive automatic update?
2. Will NanoKVM Pro have different hardware? Are there any specifications available?
3. I would be glad if @scpcom will notify us when community version will be finalized and available. Also, I hope there will be some install instructions (I assume users would need to take out SD card and install software directly). (Also, I can help writing instructions, documentation and testing).

PatrickHuetter, 2025-02-05:
Super interesting! I just ordered a few nanoKVM lite devices and i would also help if something is needed.

apalrd, 2025-02-05:
> Currently, the first generation of KVM has largely been finalized.

Do you intend to resolve the security issues in this thread for the first generation product, or are you kicking the can down the road to the Pro products?

Judging by the Github issues (including the 14 issues brought up in this thread), basically no progress has been made in reaching a bare minimum level of security, and you are already indicating that the product is 'complete'.

scpcom, 2025-02-06:
> 3. I would be glad if [@scpcom](https://github.com/scpcom) will notify us when community version will be finalized and available. Also, I hope there will be some install instructions (I assume users [...]

I added a github workflow some days ago so you can just download the licheervnano-kvm_sd.img.xz image and write it to SD like the original image:
https://github.com/scpcom/LicheeSG-Nano-Build/releases

The Debian/Ubuntu version is working too now (but still work in progress):
https://github.com/scpcom/sophgo-sg200x-debian/releases
I may publish v1.5.6 tomorrow which fixes usb gadget hid and storage (and adds the required data partition).

vikrrrr, 2025-02-06:
> There are two instructions (well, the same instruction, used twice) here that both Ghidra and Binutils are unable to disassemble, and I was unable to find any instruction using those opcodes on the [...]

> then the custom instruction presumably does something special with the data now in a5

For future reference, all these are well-known general-purpose T-Head extensions implemented in the C906 and other T-Head cores. Specifically, the ones seen here are [th.extu](https://github.com/XUANTIE-RV/thead-extension-spec/blob/master/xtheadbb/extu.adoc) and [th.srb](https://github.com/XUANTIE-RV/thead-extension-spec/blob/master/xtheadmemidx/srb.adoc). The upstream GNU toolchain (or upstream Ghidra for that matter) does not support them but you can get accurate objdump instruction listings for these by building yourself a toolchain from the XuanTie GNU toolchain sources hosted at https://github.com/XUANTIE-RV/xuantie-gnu-toolchain.

scpcom, 2025-02-06:
The cvitek toolchain is based on V2.6.1 from there, you can also find a copy here:
https://github.com/scpcom/riscv-gnu-toolchain/tree/xuantie-gnu-toolchain-v2.6.x
If you want to reproduce the toolchains in the host-tools folder you can use this scripts on Debian/Ubuntu to compile the toolchain:
https://raw.githubusercontent.com/scpcom/LicheeSG-Nano-Build/refs/heads/develop/host/build-riscv-thead-musl-toolchain.sh
https://raw.githubusercontent.com/scpcom/LicheeSG-Nano-Build/refs/heads/develop/host/build-riscv-thead-glibc-toolchain.sh

MatejKovacic, 2025-02-06:
Okeey, there is another issue. This device is basically LicheeRV Nano. And has build-in microphone.

What is a little bit weird is, that it already has Alsa tools installed. **By default.**
So you `ssh` to the device, say `amixer -Dhw:0 cset name='ADC Capture Volume 20' and then simply `arecord -Dhw:0,0 -d 3 -r 48000 -f S16_LE -t wav test.wav & > /dev/null &`.

And **bam!**, you get the file `test.wav`. Now `scp` that file to your computer and enjoy! :)

While there are certainly some useful scenarios where microphone could be very useful, this is something that needs to be documented well and - preferably - there should be an option in the GUI for enabling/disabling audio recording.

Also, an option is also to open the device and desolder microphone away:
 
[image]

On the other hand... I guess it would also be nice to have an option to play a sound through device. Basically you can use command `aplay -D hw:1,0 -f S16_LE test.wav`, but of course you would need a speaker. I tried with 8 Ohm, 0.5 W (4 cm radius), and works well.

[image]

Anyway, having a microphone and so much security issues in a receipt for disaster.

Just saying. :)

MatejKovacic, 2025-02-11:
BTW, PiKVM has [an audio support](https://docs.pikvm.org/audio/) for their devices. 
So having microphone on a board is not necessarily a bad thing. But it should be clearly documented. Better option would be to have a hardware switch to turn off mic completely. And there could also be added audio jack for speakers.

That is also why I think that my idea of plugins is not bad. Someone would be able to install audio plugin, which would allow two way communication (in case you are using NanoKVM for user support). Or this could be completely disabled.

howels, 2025-02-20:
> I added a github workflow some days ago so you can just download the licheervnano-kvm_sd.img.xz image and write it to SD like the original image: [...]

When do you plan to look at Trixie builds? I have run Trixie on an ARM laptop and it was looking ok for general use.
Also, would an apt-based image require a full image-based upgrade or could apt-based upgrades be used? Not sure what packages would need to be hosted for this.

scpcom, 2025-02-20:
Yes, you can run updates via apt, it will not break anything as far as I know.
I did not create a repository for the extra packages yet. All standard packages can be updated via dist-upgrade etc.
The latest extra packages can be found on the releases page inside licheervnano-kvm-sd_debs.tar.gz

I am sometimes testing it with Debian trixie and also see no issues. Normally I prefer Debian but this is not the offical stable release yet and I already have two different images to be tested and maintained. So switching back to Debian will happen later.

scpcom, 2025-03-07:
All extra packages for the Debian/Ubuntu images are on a deb repository now.
If you use an image with v1.5.20 or later it is already included.
For older images you can read here how to add it:
https://scpcom.github.io/

Since the packages are backward compatible I can do future releases with Debian trixie, it will start with v1.6.0.
The v1.5.x Ubuntu noble images will still get the same updates for the extra packages.
Issues section is also open for discussions:
https://github.com/scpcom/sophgo-sg200x-debian/issues
(I also think about committing pull requests to the original project by Fishwaldo but that may need some changes)

IAmBecomeDeth, 2025-04-17:
> All extra packages for the Debian/Ubuntu images are on a deb repository now. If you use an image with v1.5.20 or later it is already included. For older images you can read here how to add it: [...]

From what I can tell, you did a ton of work so thank you and don't take this as criticism. I'm a fairly technical user but have limited development experience.

The readme on GitHub basically just has a list of commands to build and install it but really NO other information aside from some chip model numbers. Based on context in this thread I'm 90% sure I understand, but can you please confirm? If everything is what I think it is...I'll be ordering the device today as the security issues were the only thing holding me back.

1. can https://github.com/scpcom/LicheeSG-Nano-Build/releases be used as a drop in replacement for the nanokvm firmware using fully open source code?
1a. does it replace the os but keep the ui/other software? either way, is it feature complete compared to stock?
2. does it no longer download the closed source library? 
3. Does it no longer submit the serial number of the device is no longer transmitted to the developer site?
4. There is a YouTuber, apalrd, who also mentioned the fact tail scale is attempting to connect even before configuring it. The implication is that it may be connecting to a developer's tail scale? Is this resolved?  or is this standard tail scale behavior if it's installed and not configured?
5. what's the difference between LicheeRV/SG-Nano-Build?
6. Your last comment seems to indicate sophgo-sg200x-debian is the Debian based version and that it (might?) be complete now that you've included the extra packages on the deb repository. Is it feature complete compared to stock?

scpcom, 2025-04-17:
1.  Yes, it is a fully open source and fully functional drop- in replacement.
 1a1. OS (buildroot 2025.02) and Kernel (5.10.235) are replaced with most recent versions (no major update for kernel to stay compatible).
 1a2. Components from the chip vendor (fsbl/u-boot/osdrv/middleware etc.) are also updated, but not the latest (2024.10.14). 
 1a3. UI (NanoKVM-Server and Web) is almost the same, built from latest sipeed sources, some components (like tailscale and stun/turn) are disabled by default but can be re-enabled.
2. No closed sourced lib will be downloaded.
3. As far as I know the code it does not (I  know much of it but do not look at every single line :-)).
4. tailscale is not running, not enabled or just short: not installed on my images
5. LicheeSG-Nano-Build is using a fully automated and reproducible build process. The github workflow creates a docker container, builds the image and publishes the images on the Releases page. But you can also clone the repository and build your own images with the same scripts (with or without docker).
6. Yes, the Debian image is feature complete now as far as I tested. If you are still missing something you can open an issue there.

Ponywka, 2025-06-05:
@scpcom What's the difference between official builds[1], [scpcom/LicheeSG-Nano-Build](https://github.com/scpcom/LicheeSG-Nano-Build) and [scpcom/sophgo-sg200x-debian](https://github.com/scpcom/sophgo-sg200x-debian)?

All three builds performed well, even though I banned NanoKVM PCIe from accessing the Internet. Why did [scpcom/LicheeSG-Nano-Build](https://github.com/scpcom/LicheeSG-Nano-Build) perform well (with hardware acceleration) even considering it can't download proprietary libraries?

[1] I tested it on the build that came with the SD card, not the ones on GitHub

scpcom, 2025-10-04:
On LicheeSG-Nano-Build you only get online updates for the NanoKVM app but not for the OS. For OS updates you need to rewrite the SD card.
So I would prefer sophgo-sg200x-debian for security because it is based on  standard Debian and you can update all packages at any time by just using apt.

ciroiriarte, 2025-10-04:
> On LicheeSG-Nano-Build you only get online updates for the NanoKVM app but not for the OS. For OS updates you need to rewrite the SD card.
> So I would prefer sophgo-sg200x-debian for security because it is based on  standard Debian and you can update all packages at any time by just using apt.
> 

is sophgo-sg200x-debian currently a full replacement for the original base firmware?, how do I add the app on top?

lowtech-guy, 2025-10-21:
> On LicheeSG-Nano-Build you only get online updates for the NanoKVM app but not for the OS. For OS updates you need to rewrite the SD card. So I would prefer sophgo-sg200x-debian for security because it is based on standard Debian and you can update all packages at any time by just using apt.

I am using squashfs as rootfs, to less chance kill my cheap sd card. Have any chance add alpinelinux disrto ?

And i found out the original sdk has freertos capture task , so we can use rtos core for image capture. without using main core resources.

scpcom, 2025-10-21:
> Sorry, but is `nanokvm-licheervnano` also opensource, so without original closed components, or not?

Yes, it is complete open source you can even view the full build process and workflows on github.

>  Does it still "phones" to China servers?

The updates are all downloaded from github.

scpcom, 2025-10-21:
> And i found out the original sdk has freertos capture task , so we can use rtos core for image capture. without using main core resources.

Image capture and video encoding/decoding is always done on the "rtos core", there is a component named ISP that is running on this core, the middleware used by MaixCDK and NanoKVM controls this ISP and gets/puts the image from/to it.