sipeed/NanoKVM #272: Keyboard not working

raphaeldias85, 2025-01-21:
Hello,

I attached the USB from outsite or inside, same issue: no keyboard.

[ 5208.624737] mce: [Hardware Error]: Machine check events logged
[ 5261.708806] usb 3-13: USB disconnect, device number 23
[ 5262.955571] usb 3-13: new high-speed USB device number 24 using xhci_hcd
[ 5268.574053] usb 3-13: New USB device found, idVendor=3346, idProduct=1009, bcdDevice= 5.10
[ 5268.574065] usb 3-13: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5268.574070] usb 3-13: Manufacturer: sipeed
[ 5273.693978] usb 3-13: can't set config #1, error -110
[ 5371.517110] usb 3-13: USB disconnect, device number 24
[ 5379.256350] usb 3-13: new high-speed USB device number 25 using xhci_hcd
[ 5384.797402] usb 3-13: unable to read config index 0 descriptor/all
[ 5384.797414] usb 3-13: can't read configurations, error -110
[ 5384.911324] usb 3-13: new high-speed USB device number 26 using xhci_hcd
[ 5390.429392] usb 3-13: unable to read config index 0 descriptor/all
[ 5390.429403] usb 3-13: can't read configurations, error -110
[ 5390.429520] usb usb3-port13: attempt power cycle
[ 5390.817321] usb 3-13: new high-speed USB device number 27 using xhci_hcd
[ 5396.064331] usb 3-13: unable to read config index 0 descriptor/all
[ 5396.064344] usb 3-13: can't read configurations, error -110
[ 5396.180259] usb 3-13: new high-speed USB device number 28 using xhci_hcd
[ 5401.693370] usb 3-13: unable to read config index 0 descriptor/all
[ 5401.693381] usb 3-13: can't read configurations, error -110
[ 5401.693487] usb usb3-port13: unable to enumerate USB device
[ 5870.745650] mce: [Hardware Error]: Machine check events logged
[ 6103.735920] usb 3-13: new high-speed USB device number 29 using xhci_hcd
[ 6109.275973] usb 3-13: unable to read config index 0 descriptor/all
[ 6109.275987] usb 3-13: can't read configurations, error -110
[ 6109.389915] usb 3-13: new high-speed USB device number 30 using xhci_hcd
[ 6114.907983] usb 3-13: unable to read config index 0 descriptor/all
[ 6114.907994] usb 3-13: can't read configurations, error -110
[ 6114.908109] usb usb3-port13: attempt power cycle
[ 6115.287900] usb 3-13: new high-speed USB device number 31 using xhci_hcd
[ 6120.539950] usb 3-13: unable to read config index 0 descriptor/all
[ 6120.539961] usb 3-13: can't read configurations, error -110
[ 6120.653890] usb 3-13: new high-speed USB device number 32 using xhci_hcd
[ 6126.171935] usb 3-13: unable to read config index 0 descriptor/all
[ 6126.171946] usb 3-13: can't read configurations, error -110
[ 6126.172057] usb usb3-port13: unable to enumerate USB device

Z2Z-GuGu, 2025-01-22:
It seems from the logs that the issue might be caused by a poor connection with the USB-C (it disconnects after a while). Please try replacing the USB cable. (I also want to complain about the quality of the included cable, which forced me to use my own high-quality USB cable during development.)

Z2Z-GuGu, 2025-01-22:
If the keyboard, mouse, virtual USB drive, and other devices cannot be used simultaneously, it is likely due to this issue. Additionally, you can update to the latest version and try clicking 'Reset HID' on the web.

raphaeldias85, 2025-01-22:
I tried to use the two cables included and other I own.

I tried to use the internal pins into the motherboard, same scenario.

Z2Z-GuGu, 2025-01-22:
May I ask what host device is connected to the NanoKVM? Additionally, is it only the keyboard that is not working, or are all devices emulated by the NanoKVM, such as the mouse, also not functioning?

raphaeldias85, 2025-01-22:
Sure

I was using Geekworm-X650 PCIe version in the same server and no issues. I
even used the same USB-C cable for a try.

Anyway, I am using a X99 F8D Plus with dual Xeon v4 E5-2696 and 256GB ECC
RAM.

I also tried:

touch /boot/BIOS && restart

And hit reset HID a couple of times.

I am only trying to use the keyboard.

Z2Z-GuGu, 2025-01-22:
As far as I know, some operating systems may not allow HID devices and mass storage devices to be combined into the same USB device. You might try disabling the virtual USB drive and RNDIS in the web settings.

Z2Z-GuGu, 2025-01-22:
Hmm... the likelihood is low. I took a closer look at the log above and found that the host didn't even retrieve the USB descriptor for NanoKVM. This might be related to the gadget service in the NanoKVM system. Please try connecting the USB HID interface to another computer to see if the USB device appears. If it still doesn't work, you may need to re-flash the NanoKVM system according to the instructions [here](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html) (note that the new system will automatically fetch the latest applications and download them upon first power-up, which may take some time).

CRCinAU, 2025-01-22:
I'm having a hell of a time using the USB Keyboard too.... If I reboot the NanoKVM, the keyboard will work for a little bit of time, then stop functioning again.

raphaeldias85, 2025-02-21:
I guess we will all have this kvm as a brick. Waste of money. I got 2
geerworm KVM working just fine.

This is a dead thread it seems. The dev does not care.

CaneTLOTW, 2025-02-21:
PiKVM is working finr on same system with sames cables. Issues definatly related to NanoKVM.

Z2Z-GuGu, 2025-03-26:
We recently updated to version 2.2.5, which introduces 'HID Only' mode to improve HID compatibility issues. Please select this mode from the mouse icon after the update. We have reproduced your testing results on multiple computers, and using the new mode resolved the problem. I can't guarantee that this update will work for everyone's systems, but we look forward to receiving your testing results after the update. Thank you!

roondar, 2025-04-23:
Same here with  "HID Only" mode enabled: 
```
[    1.427007] usb 1-12: new full-speed USB device number 4 using xhci_hcd
[    1.543736] usb 1-12: device descriptor read/64, error -32
[    1.763728] usb 1-12: device descriptor read/64, error -32
[    1.867097] usb usb1-port12: attempt power cycle
[    2.253763] usb 1-12: new full-speed USB device number 5 using xhci_hcd
[    2.267756] usb 1-12: device descriptor read/8, error -32
[    2.384414] usb 1-12: device descriptor read/8, error -32
[    2.603713] usb 1-12: new full-speed USB device number 6 using xhci_hcd
[    2.617757] usb 1-12: device descriptor read/8, error -32
[    2.734381] usb 1-12: device descriptor read/8, error -32
[    2.837125] usb usb1-port12: unable to enumerate USB device
[    2.846804] usbcore: registered new interface driver usbhid
[    2.846813] usbhid: USB HID core driver
```

ruimo, 2025-09-10:
Same here. Any update on this? The followings do not work:

- Change USB cable
- Use USB hub
- Turn off virtual disk/virtual network

```
[    4.237837] usb 1-8: new high-speed USB device number 12 using xhci_hcd
[    9.573988] usb 1-8: device descriptor read/64, error -110
[   25.445994] usb 1-8: device descriptor read/64, error -110
[   25.927836] usb 1-8: new high-speed USB device number 13 using xhci_hcd
[   31.077873] usb 1-8: device descriptor read/64, error -110
[   46.949857] usb 1-8: device descriptor read/64, error -110
[   47.051937] usb usb1-port8: attempt power cycle
[   47.115814] usb 1-2.3: new full-speed USB device number 14 using xhci_hcd
[   47.196889] usb 1-2.3: New USB device found, idVendor=0582, idProduct=0099, bcdDevice= 1.00
[   47.196900] usb 1-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   47.196904] usb 1-2.3: Product: UM-2
[   47.196907] usb 1-2.3: Manufacturer: EDIROL
[   47.371799] usb 1-4.4: new high-speed USB device number 15 using xhci_hcd
[   47.563139] usb 1-4.4: New USB device found, idVendor=291a, idProduct=3361, bcdDevice= 3.10
[   47.563150] usb 1-4.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   47.563153] usb 1-4.4: Product: Anker PowerConf C300
[   47.563156] usb 1-4.4: Manufacturer: Anker
[   47.563159] usb 1-4.4: SerialNumber: AR2MXP0B24401260
[   47.636811] usb 1-4.1.4: new high-speed USB device number 16 using xhci_hcd
[   47.741769] usb 1-4.1.4: New USB device found, idVendor=0bda, idProduct=5411, bcdDevice= 1.23
[   47.741782] usb 1-4.1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   47.741786] usb 1-4.1.4: Product: 4-Port USB 2.0 Hub
[   47.741789] usb 1-4.1.4: Manufacturer: Generic
[   47.743804] hub 1-4.1.4:1.0: USB hub found
[   47.744871] hub 1-4.1.4:1.0: 4 ports detected
[   47.796711] usb 1-4.2.4: new full-speed USB device number 17 using xhci_hcd
[   47.875171] usb 1-4.2.4: New USB device found, idVendor=04f3, idProduct=0c3d, bcdDevice= 1.65
[   47.875177] usb 1-4.2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   47.875180] usb 1-4.2.4: Product: ELAN:Fingerprint
[   47.875183] usb 1-4.2.4: Manufacturer: ELAN
[   48.191828] usb 1-8: new high-speed USB device number 18 using xhci_hcd
[   53.491876] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[   59.123855] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[   59.331831] usb 1-8: device not accepting address 18, error -62
[   59.711836] usb 1-8: new high-speed USB device number 19 using xhci_hcd
[   64.755986] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[   70.387972] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[   70.595827] usb 1-8: device not accepting address 19, error -62
[   70.596069] usb usb1-port8: unable to enumerate USB device
```

Mifody-07, 2026-10-29:
I’m not certain if other versions of the NanoKVM have the same grounding issue, but you should still try grounding both the NanoKVM and the network switch. look at my post in this https://github.com/sipeed/NanoKVM/issues/26#issuecomment-6373769940