sipeed/NanoKVM #274: New NanoKVM, version 2.1.5?

alexisfrjp, 2025-01-23:
I just received my Sipeed NanoKVM and I'm suprised to see the version 2.1.5 whereas github's latest release version is 1.5.0.

What am I missing?

[image]

[image]

Thanks!

matdibu, 2025-01-23:
I just got mine delivered with 2.1.4, and it updated to 2.1.5 just fine

it was downloaded from the company's CDN, not GitHub releases

alexisfrjp, 2025-01-23:
I am not asking whether I do the update but why the version on my device is 1 major above github's. Where does it come from? I won't use a firmware I can't compile myself.

alexisfrjp, 2025-01-23:
> I imagine the version mismatch is a difference between the image version vs. the application version.
> 
> On my end I have 2.1.2 and the update isn't working. It starts and then nothing happens.

Whefe is the repo for the app then?

muness, 2025-01-23:
> I imagine the version mismatch is a difference between the image version vs. the application version.
> 
> On my end I have 2.1.2 and the update isn't working. It starts and then nothing happens. 

Worked eventually. I think I wasn't waiting long enough. Doing it on the command line at showed progress where I could see the slow download work. After the update only port 80 works, 443/https doesn't appear to be listening to requests.

ucki007, 2025-01-23:
Hi everyone,

already discussed that in #191 .

Imageversion (current: 1.3.0) is the firmware (so to speak the OS, that runs your NanoKVM).
Application Version (current: 2.1.5) is the application that provides you with things like the Website etc. running on the device.

I don't know, if both are mixed in the repo here or if there is another (maybe non-public) repo, that provides the application

alexisfrjp, 2025-01-23:
I see the versionning transaprency hit other people too.
Thanks! I'll wait for the answer of where is the source of the app.

closed 2025-01-23 by alexisfrjp

alexisfrjp, 2025-01-23:
Ok, I checked and 2.1.5 seems to be the system.

https://github.com/sipeed/NanoKVM-System