sipeed/NanoKVM #276: Strange Video Freezing

li20034, 2025-01-24:
I'm getting some strange video freezing every few seconds in H.264 mode. It's smooth for about 10 seconds, and freezes for 2-3 seconds. In MJPEG mode, everything is just very choppy. 

My network connection is perfectly fine, ping times are well under 10ms. Also, it used to work fine in December, before I updated to 2.1.5 earlier today, I think. Here's the version details:
```
Image Version v1.3.0
Application Version 2.1.5
```

dmesg shows a lot of messages like:
```
[  677.118828] configfs-gadget gadget: usb_gadget_wakeup
[  677.118936] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
[  677.118949] configfs-gadget gadget: usb_gadget_wakeup
[  677.182376] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
[  677.182393] configfs-gadget gadget: usb_gadget_wakeup
[  677.182478] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
[  677.182489] configfs-gadget gadget: usb_gadget_wakeup
```

Is there a way to downgrade back to 2.1.2 or 2.1.3 to test it?

CRCinAU, 2025-01-24:
I've been getting all sorts of issues like this too - including network connectivity problems...

I seem to get a somewhat lagged display when using MJPEG - but at least I get an output... The h264 stream normally shows a blank screen...

I don't know what has happened over the last few months - but I've had nothing but trouble with all of my NanoKVMs after updating them...

uwuclara, 2025-01-25:
Thank god I'm not crazy. I just got NanoKVM and been having the same issue. H264 a lot of times black screen and when it does load, it seems to freeze after few seconds and unfreezes on repeat. MJPEG always loads but it's rly laggy and even showing around 10 FPS.

li20034, 2025-02-16:
Seems like app version 2.1.6 fixes this for me

CRCinAU, 2025-02-16:
Updated one NanoKVM Full to 2.1.6 - no difference. MJPEG gets a picture, h264 doesn't show anything.

uwuclara, 2025-02-16:
Can confirm. It only seems to be doing it for me on my Chuwi larkbox mini PC, other devices fine.

b4nds, 2025-03-04:
Im having the same problem. I have mine connected to a Mini PC which is running Ubuntu Desktop. h264 doesnt show any output and MJPEG is just a frozen screen but I see output. Is there any fix or any news ?

kariudo, 2025-03-18:
Seeing the same thing on 1.4.0 & 2.2.2

I was previously able to get h264, but after the first day, it now only will show an image for MJPEG.

- MJEG works (with a couple seconds of latency)
- H264 gives a black screen.
- Resets/reboots do not change result
- Rollback to 2.2.0 does not change result
- Re-plugs of everything do not change result
- Different HDMI and USB C-A cables do not change result
- Running `/etc/init.d/S95nanokvm restart_stream` does not change result

`dmesg` shows the same as OP above:

```
[ 7914.401779] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
[ 7914.401794] configfs-gadget gadget: usb_gadget_wakeup
[ 7914.441573] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
[ 7914.441589] configfs-gadget gadget: usb_gadget_wakeup
[ 7914.452503] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
[ 7914.452518] configfs-gadget gadget: usb_gadget_wakeup
[ 7914.461550] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
[ 7914.461566] configfs-gadget gadget: usb_gadget_wakeup
```

```sh
# cat /proc/cvitek/vi_dbg
[VI BE_Dbg_Info]
VIPreBEDoneSts          :0x27           VIPreBEDmaIdleStatus    :0x7
[VI Post_Dbg_Info]
VIIspTopStatus          :0x3fe
[VI DMA_Dbg_Info]
VIWdma0ErrStatus        :0x3000000      VIWdma0IdleStatus       :0xfffffffe
VIWdma1ErrStatus        :0x3000000      VIWdma1IdleStatus       :0xffffffff
VIRdmaErrStatus         :0x3000000      VIRdmaIdleStatus        :0xffffffff
[VI ISP_PIPE_A FE_Dbg_Info]
VIPreFERawDbgSts        :0x3f0          VIPreFEDbgInfo          :0x3e
[VI ISP_PIPE_A]
VIOutImgWidth           :1920
VIOutImgHeight          :1080
VIInImgWidth            :1920
VIInImgHeight           :1080
VIDevFPS                :  52
VIFPS                   :  60
VISofCh0Cnt             :424057
VIPreFECh0Cnt           :504168
VIPreBECh0Cnt           :   0
VIPostCnt               :   0
VIDropCnt               :   0
VIDumpCnt               :   0
[VI ISP_PIPE_A Csi_Dbg_Info]
VICsiIntStatus0         :0x6
VICsiIntStatus1         :0x0
VICsiCh0Dbg             :0x0
VICsiCh1Dbg             :0x0
VICsiOverFlowCnt        :   0
VICsiCh0WidthGTCnt      :   0
VICsiCh0WidthLSCnt      :   0
VICsiCh0HeightGTCnt     :   1
VICsiCh0HeightLSCnt     :   1
```

```sh
# cat /etc/kvm/hdmi_version
ux
```

IceDragonix, 2025-04-11:
Exactly same issue as described by @kariudo. 

MJPEG is working as expected. No video using H.264.

civictd04, 2025-04-23:
There's a known design flaw with the NanoKVM where the Ethernet can be flaky. The recommend fix by another user was to use a shielded cat cable with metal connectors. You can buy a cat7 or cat8 cable and it should provide better stable network connection. On Amazon you can buy a 3ft cat8 cable for about $6.

mpajer, 2025-04-24:
Thank you very much @civictd04 🥇, I tried multiple cables but now I tried cat8 cable and it really helped 👍. Connection is stable and fast even on H.264

mpajer, 2025-04-26:
UPDATE:

So a better cable was just a temporary solution, the problem appeared again. But I finally figured out what the issue was in my case. It was my Asus router that repeatedly forced a gigabit connection. When the nanokvm was connected directly to the router, the connection was unstable, the IP address on the OLED display kept flashing, and the OLED display never turned off, even when the timeout was set to 15 sec. But when I connected it to a TP-Link switch that is connected directly to the router, all problems disappeared. The connection is stable, and the OLED display turned off properly.

civictd04, 2025-04-26:
I have heard that putting a quality switch between the nano and the router can help too.