sipeed/NanoKVM #277: Update Get Versione failed

Luci4n89, 2025-01-24:
Hi

I'm trying to update from 2.1.3 to 2.1.5 but it fails every time.

[image]

ciroiriarte, 2025-01-24:
Hello!, 

I see the same error going from 2.0.9 to 2.1.5, also it's odd versions don't match https://github.com/sipeed/NanoKVM/releases, am I missing something?.

Luci4n89, 2025-01-24:
> Hello!,
> 
> I see the same error going from 2.0.9 to 2.1.5, also it's odd versions don't match https://github.com/sipeed/NanoKVM/releases, am I missing something?.

Hi

I updated to 1.3.0 (I wrote the image to the SD card) and then I was able to update to 2.1.5.

ciroiriarte, 2025-01-25:
This the same, had to redownload the ISO files I lost from /data.

Works for the time being, super ugly fix if device is remote (unreachable)

wj-xiao, 2025-02-18:
The latest version supports offline updates. You can manually update the [Application](https://github.com/sipeed/NanoKVM/releases/tag/2.2.0) refer to the [README.md](https://github.com/sipeed/NanoKVM/blob/main/server/README.md#manually-update).