sipeed/NanoKVM #280: Please add reboot nano-kvm button on the web interface

yen360, 2025-01-26:
[image]

Please add reboot the nano-kvm button here

sizkein, 2025-01-27:
There is no button.
The only way is to press the terminal button at the top (http://your-ip/#terminal) and connect and reboot.
However, when nano-kvm had an error before, I couldn't connect to either the webpage or the console, so I physically turned it off and on.

Blackheat45, 2025-02-01:
@sizkein One of the worst things is to often I'm having to physically unplug and replug. I'm not sure why. But when I can't physically then I'm just SOL. Do you know why this happens? It's so random on when it locks up for me. I was hoping the new tailscale ram setting would fix it but still every so often locks up.

kam821, 2025-02-06:
Temporary workaround: send a new script named e.g. reboot_nanokvm.sh that contains:
```
#!/bin/bash

echo "Rebooting NanoKVM..."
reboot

exit 0
```

Blackheat45, 2025-02-08:
> Temporary workaround: upload a script named e.g. reboot_nanokvm.sh that contains:
> 
> ```
> #!/bin/bash
> 
> echo "Rebooting NanoKVM..."
> reboot
> 
> exit 0
> ```

Do you know of a way I can put this on a timer? Like using watch command so it runs the reboot every hour for example? Not sure how to upload to the nano and make it work either. Having it on a timer would solve the issue it normally takes some time to lock up. Even having a 10min restart would be better so less likely to have it ever lock up. Thanks for the help.

olekkut, 2025-02-10:
> Do you know of a way I can put this on a timer? Like using watch command so it runs the reboot every hour for example? Not sure how to upload to the nano and make it work either. Having it on a [...]

@Blackheat45 I can’t test this on my NanoKVM right now, but I’ll try it when I get the chance. In the meantime, you can set up an automatic reboot using cron instead of watch. Here’s how:

Upload a script named reboot_nanokvm.sh with this content:

```
#!/bin/bash
echo "Rebooting NanoKVM..."
reboot
exit 0
```
Make it executable:
```
chmod +x reboot_nanokvm.sh
```
Then, add a cron job to run it automatically. Open the crontab editor:
```
crontab -e
```
Add this line to reboot every 10 minutes:
```
*/10 * * * * /path/to/reboot_nanokvm.sh
```
Or if you prefer an hourly reboot:
```
0 * * * * /path/to/reboot_nanokvm.sh
```
Check if the cron job was added:
```
crontab -l
```

prolife86, 2025-03-19:
> [@Blackheat45](https://github.com/Blackheat45) I can’t test this on my NanoKVM right now, but I’ll try it when I get the chance. In the meantime, you can set up an automatic reboot using cron [...]

Good plan, but it does not work as there is no crontab available:
`
# crontab -e
crontab: can't change directory to '/var/spool/cron/crontabs': No such file or directory
`
https://github.com/sipeed/NanoKVM/issues/211

wj-xiao, 2025-04-08:
Added in version [2.2.2](https://github.com/sipeed/NanoKVM/blob/main/CHANGELOG.md#222-58d5ab2-2025-03-11).

closed 2025-04-08 by wj-xiao