sipeed/NanoKVM #282: Bad unit?

Treebear69, 2025-01-26:
Got the full KVM today, plugged it in and got a display, but no IP.  After a bit of digging, I found that I should update the image.  Since flashing to the latest image, all I can get is "Please connect to network to update app" on the screen which never goes away.  When I can connect to the web UI through USB, I can't update the software, or see any HDMI input. Seriously thinking about just sending back, after reading how many issues there are with these.

ucki007, 2025-01-27:
Hi @Treebear69 , 

that may be caused by the application repository not being available.

As the first task of a factory default NanoKVM (which should be the case for a new device and for devices with a freshly flashed SD card) there is the update of the application. For me sometimes the connectivity to the CDN is very bad and if so there is no or very slow progress with the update. As you have no progress bar or anything like that, it may look like the device does nothing.

Im not saying that it should be like that, but nobody is perfect...

Treebear69, 2025-01-28:
@ucki007 
Is there a bug in the current version that stops the unit from getting an IP? I have DHCP on and my switch is 100M compatible. The unit was given an IP and was working fine, but it froze up and when I rebooted it, it isn’t grabbing an IP anymore.

tishcufksips, 2025-01-31:
Did you try with HDMI and USB cable disconnected? I was having issues with grounding and was only getting IP with USB connected to an AC adapter (without PC connection).
What helped in my case was an FTP cable with shielded connectors (check if there is continuity on both ends).
Hope it helps.

ucki007, 2025-01-31:
> [@ucki007](https://github.com/ucki007) Is there a bug in the current version that stops the unit from getting an IP? I have DHCP on and my switch is 100M compatible. The unit was given an IP and was working fine, but it froze up and when I rebooted it, it isn’t grabbing an IP anymore.

I have no idea, if there is any bug known in this direction. For me, I have no issues regarding DHCP / general connectivity (except that currently DNS servers cannot be set using DHCP)

dotX12, 2025-02-01:
see: https://github.com/sipeed/NanoKVM/issues/223#issuecomment-2629114317

Treebear69, 2025-02-03:
> see: [#223 (comment)](https://github.com/sipeed/NanoKVM/issues/223#issuecomment-2629114317)

I was using a shielded cable and still having issues. I ended up returning the unit before my return window ended.  Too many issues made it not worth keeping.

CRCinAU, 2025-02-04:
> I was using a shielded cable and still having issues. I ended up returning the unit before my return window ended. Too many issues made it not worth keeping.

Given all the issues I'm having, I kinda wish I did the same - but I got them shipped in August last year... Got 5 of the full versions, almost all of them don't even get link unless I power them from a USB Battery bank - which makes them pretty much useless.

Tried many USB power supplies, phone chargers etc, nothing except the battery pack seems to be reliable.

I'm outside of my return window, unless I try to claim the warranty / not fit for purpose via AliExpress - which I probably should have done quite some time ago...