sipeed/NanoKVM #283: HTTPS not working

danb35, 2025-01-28:
HTTPS had been working just fine on my NanoKVM lite unit (it still works on my two units running 2.1.1), but stopped working with the update to, IIRC, 2.1.2, and it continues to not work under 2.1.5.  The interface is instead served over HTTP.  The server.yaml file is set properly:
```yaml
# cat /etc/kvm/server.yaml
protocol: https
port:
  http: 80 
  https: 443
cert:
  crt: /etc/ssl/certs/fullchain.pem
  key: /etc/ssl/private/privkey.pem
```
The cert and key files both exist, and they contain a valid Let's Encrypt cert and private key.  But nonetheless the interface is served over HTTP, not HTTPS.  What needs to change in order to effect this?

danb35, 2025-02-06:
Thanks to Apalard on YT for finding the problem: https://github.com/sipeed/NanoKVM/commit/5a39562f2d32695933f4e7e86866136236cc9903#diff-19136d80005b994ac9a98d8d280b36fc5e6a5be77cdf68eac6cafc55e029d04c for some reason changed the required key.  To enable HTTPS, the first line now needs to read `proto: https`, not `protocol:`.  No idea why this undocumented change was made.  But I left a note on the wiki page.

closed 2025-02-06 by danb35

bilogic, 2025-02-18:
how did you get your keys onto the device, and keep it up to date?

bilogic, 2025-02-18:
I see, that would be push via SSH.. hmm ok thanks!