sipeed/NanoKVM #287: Picture in Quick Start Guide is incorrect

KuroSetsuna29, 2025-01-29:
In the quick guide documentation for [Wiring](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/quick_start.html#Wiring), the middle picture is suppose to show "only the power button is connected here", however this conflicts with the wiring guide on the right picture. Note, the orientation is flipped between the middle and right. Which means it is actually connected between HDD LED- & RST SW? Unless the right diagram is incorrect?

[image]

wj-xiao, 2025-05-28:
Thanks for your reminder. The picture has been updated.

KuroSetsuna29, 2025-05-28:
@wj-xiao I think the middle picture is still wrong compared to the text (only the power button is connected here). That appears to be the RST SW (not power switch)

[image]

It becomes appear if I rotate the right between to match the same orientation.

[image]

Z2Z-GuGu, 2025-05-28:
@KuroSetsuna29 I'm very sorry, the last image is still incorrect. I will make the necessary changes as soon as possible. Thank you for your reminder

KuroSetsuna29, 2025-05-28:
Is it the last image that is incorrect? Because that matches the silkscreen labels on the back side. I thought it was the middle image that was the issue.