sipeed/NanoKVM #289: Initial setup fails on networks with China geo-blocked

elliotmatson, 2025-01-29:
New units will get an IP and briefly show the dashboard, but as soon as update-nanokvm.py runs the dashboard shuts down. The update script also fails and loops continuously because it can't access maixvision.sipeed.com. This will likely be addressed in #270, but since geo-blocking China is a default security rule in a large number of SMB/enterprise environments, this seemed worth noting

wj-xiao, 2025-02-18:
The latest version removes the maixvision.sipeed.com dependency and supports offline updates.

You can now manually update the [Application](https://github.com/sipeed/NanoKVM/releases/tag/2.2.0) (see [README.md](https://github.com/sipeed/NanoKVM/blob/main/server/README.md#manually-update)).

Or flash to the latest [Image](https://github.com/sipeed/NanoKVM/releases/tag/v1.4.0) (see [Flashing](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html)).

elliotmatson, 2025-02-18:
Thanks so much for for y'alls hard work on this last update! I'll close this issue

closed 2025-02-18 by elliotmatson