sipeed/NanoKVM #291: NanoKVM PCIe - no display, no IP

elf55, 2025-02-01:
I have a new NanoKVM PCIe that just arrived and I've installed in single lane PCIe slot in a ZimaCube pro and connected the HDMI, USB, PoE Ethernet, and reset, power LED, and HDD pins.   The NanoKVM powers on and flashes the green LED on the back several times.  The Ethernet switch shows that link is up.  It never requests in IP address and the display panel is dark.  I have pressed the reset switch and held it down for two seconds without results.  The Ethernet link goes down and then comes back up and the green LED blinks a few times.  

I have put in a new SD card with firmware version 1.3.0 and see no different result.  

Can someone confirm that the device is broken or is there some other reason for it to be almost dead?

iceton, 2025-03-11:
Can you feel your reset switch depress with a small click feeling? I encountered the same thing and my boot button was stuck. I removed the bracket and it worked as expected. The hole for the boot button seems to be too small so I will drill it out and reattach.

EnzoAlbornoz, 2025-04-06:
> Can you feel your reset switch depress with a small click feeling? I encountered the same thing and my boot button was stuck. I removed the bracket and it worked as expected. The hole for the boot button seems to be too small so I will drill it out and reattach.

Thank you very much! This helped me a lot. I managed to 3D-print an alternative bracket for it. I published it in Printables with an Open Source license:
https://www.printables.com/model/1256646-sipeed-nanokvm-pcie-bracket-full-height