sipeed/NanoKVM #295: Do you have any plans to introduce MFA when you log in?

wildcong, 2025-02-04:
This is a security-related question.
I think it will help with security when logging in to MFA, do you have any plans to introduce it?

stuartm, 2025-02-04:
Bonus points if you use WebAuthn (W3C Standard, supported by all major browsers) and _very_ simple to implement. https://auth0.com/blog/webauthn-a-short-introduction/

TuningYourCode, 2025-02-09:
WebAuthn would be nice but requires a https connection. As far as this is currently not the case? I'm also unsure if self signed certificates are allowed by the major browsers for WebAuthn.

TOTP on the other side should be pretty doable while locking at the code.

CRCinAU, 2025-02-13:
> TOTP on the other side should be pretty doable while locking at the code.

Without an RTC?

wj-xiao, 2025-02-19:
Yes, I'll add this to the roadmap.

TuningYourCode, 2025-02-19:
> > TOTP on the other side should be pretty doable while locking at the code.
> 
> Without an RTC?
> 
> Otherwise you hit the real possibility of not having internet, booting the KVM and not being able to log in at all...

sorry, didn't thought about that. NTP might be a solution but only if internet connection is present.

HOTP might be an option and has also an RFC (it's counter based instead of time). Of course Webauthn might be nice but then we first need to solve the missing TLS/HTTPS.

CRCinAU, 2025-02-19:
> sorry, didn't thought about that. NTP might be a solution but only if internet connection is present.

Yeah - TOTP won't work if booted with no ability to sync time - so when you'd need a KVM the most, you wouldn't be able to use TOTP.

> HOTP might be an option and has also an RFC (it's counter based instead of time). Of course Webauthn might be nice but then we first need to solve the missing TLS/HTTPS.

Potentially, but I haven't seen anything actually use HOTP for a *very* long time.

Webauthn would require both a working DNS infrastructure (it must use a domain name), and TLS certificates to match.

I personally think the majority of implementations of MFA would be overkill for this device - and the majority have the potential to lock you out when you'd need the KVM the most - ie when having problems where you need a console.

stuartm, 2025-02-19:
Having it as an optional feature would alleviate any concerns. Personally I already have an FQDN for my NanoKVM and a TLS cert is a trivial thing these days with LetsEncrypt so IMHO those shouldn't be blockers to the implementation of WebAuthn.

TLS seems like a pretty important thing too, if only to protect passwords in single factor auth.

Personally, I'd rather be locked out of my KVM than leave it exposed, or more accurately the server it's connected to exposed. Call it an abundance of caution, even if I'm not putting my KVM directly onto the internet and I'm using tailscale to connect, I still prefer to have some depth to my security - particularly in time when there are so many devices already in the LAN that cannot be entirely trusted.

CRCinAU, 2025-02-19:
> Having it as an optional feature would alleviate any concerns.

However having a method to bypass it when it breaks is just as good as not having it at all.

stuartm, 2025-02-19:
> > Having it as an optional feature would alleviate any concerns.
> 
> However having a method to bypass it when it breaks is just as good as not having it at all.

Optionally **enabled**,  not optional when logging in once it has been enabled.

CRCinAU, 2025-02-19:
> > > Having it as an optional feature would alleviate any concerns.
> > 
> > 
> > However having a method to bypass it when it breaks is just as good as not having it at all.
> 
> Optionally **enabled**, not optional when logging in once it has been enabled.

Sure - and, say your DNS dies. Or your cert expires with no internet to renew. How do you recover your device if using WebAuthn without disassembling it to factory reset it?

I kinda think you guys underestimate how these ways kill what is essentially a 'last resort recovery' for problems.

Sure, maybe *you* like it, but I wouldn't even call it a good idea for a default installation.

I mean, the good thing about open source is you can fork the project and then break it to your hearts content, but the majority of KVM users (not just this project) need them to be fully operational in even minimal network conditions.

stuartm, 2025-02-19:
> Sure - and, say your DNS dies. Or your cert expires with no internet to renew. How do you recover your device if using WebAuthn without disassembling it to factory reset it?

SSH ...

Or as you say factory reset.

Why are you searching so hard for some reason not to improve security of a device I own? I mean if I don't have internet access etc surely I have bigger issues than not being able to access the KVM?

If users manage to lock themselves out of it, that's their problem, what does it matter to you? If you don't want the option, don't enable it?

> I kinda think you guys underestimate how these ways kill what is essentially a 'last resort recovery' for problems.

It's only the last resort if you don't have physical access to the machine it's connected to. By your own argument, if you don't have internet and not your not physically local you're already stuffed?

> Sure, maybe _you_ like it, but I wouldn't even call it a good idea for an option to be selectable.

And maybe _you_ would not like it, so just don't use it?!?

> I mean, the good thing about open source is you can fork the project and then break it to your hearts content, but the majority of KVM users (not just this project) need them to be fully operational in even minimal network conditions.

How is it you think you speak for the majority? I mean frankly speaking every large organisation I've worked with in recent years wouldn't even allow a device with just password authentication anywhere near their networks.

CRCinAU, 2025-02-19:
> SSH ...

Which MFA should be used to secure SSH?

mikkorantalainen, 2025-03-07:
> TOTP on the other side should be pretty doable while locking at the code.

I think a device without RTC should use HOTP instead. The best method would be to support both. Then TOTP could be used when ntp is working and HOTP would be reserved for recovery situations only which would also help avoiding out-of-sync errors for HOTP sequence, especially if you have multiple admins that need to connect to the device.