sipeed/NanoKVM #296: HDMI Audio not playing

joetretter, 2025-02-05:
Hello,

I assumed that audio would be played if I change my computer's audio to HDMI. 
It doesn't work for me though and I'm wondering if it's supposed to work at all.
If this is not yet a feature, is the hardware capable of doing it? 

Thanks

fallenleavesgocrunch, 2025-02-06:
It doesn't support HDMI Audio

joetretter, 2025-02-06:
> It doesn't support HDMI Audio

do you know if this is a hardware limitation (in which case this can't be fixed by software).

azbesthu, 2025-02-14:
Video is converted to CSI, so I guess it is a HW limitation, because that is a video port. On other devices audio usually goes to i2s port from the converter.

samascience, 2025-02-18:
this will not be possible.

samascience, 2025-02-18:
> Video is converted to CSI camera input, so I guess it is a HW limitation, because that is a video port. On other devices audio usually goes to i2s port from the converter. Not sure if the converter board has audio output and LicheeRV Nano has i2s input

seems like there is a audio input, but I highly doubt it has been implemented in this setup !