sipeed/NanoKVM #297: NanoKVM does not support IPv6

apalrd, 2025-02-05:
When connected to a dual-stack network, an IPv6 address is not assigned.

ValdikSS, 2025-02-05:
That's because it acts as a router, which automatically disables receiving IPv6 Router Advertisement
https://github.com/sipeed/NanoKVM/issues/197

apalrd, 2025-02-05:
It doesn't matter 'why' it doesn't support IPv6.

It does not support IPv6, and until that is fixed, this is an issue.

ValdikSS, 2025-02-05:
Nobody's arguing.

gshpychka, 2025-02-18:
> It doesn't matter 'why' it doesn't support IPv6.

It does, since this is a GitHub issue, where debugging the root cause and any info related to that would be useful.

wj-xiao, 2025-02-19:
Image [v1.4.0](https://github.com/sipeed/NanoKVM/releases/tag/v1.4.0) has IPv6 enabled by default.

However, starting Tailscale disables IPv6 permanently.  We will fix it.

jpatriarca, 2025-06-26:
Any updates on this IPv6 support with tailscale @wj-xiao? Is it on the roadmap for upcoming versions?
Also, to add, there will be support for the manual IPv6 configuration and an option to ignore RA?
Thank you

RipperFox, 2025-07-21:
> That's because it acts as a router, which automatically disables receiving IPv6 Router Advertisement [#197](https://github.com/sipeed/NanoKVM/issues/197)

It's possible to configure the kernel to accept RAs when routing is enabled via a simple sysctl setting:
`net.ipv6.conf.all.accept_ra=2`

See https://www.kernel.org/doc/Documentation/networking/ip-sysctl.txt

RipperFox, 2025-08-21:
> ## Workaround: IPv6 SLAAC Working Solution **(without IPv6 forwarding)**

Great! But did you even read my comment above yours, which provides a workaround WITH IPv6 forwarding?
Did you try putting `net.ipv6.conf.all.accept_ra=2` in /etc/sysctl.conf?

nbanks, 2025-08-22:
In my case, I had /etc/sysctl.d/99-tailscale.conf which I moved, and then added a file:
/etc/sysctl.d/31-ipv6.conf with the one line `net.ipv6.conf.all.accept_ra=2`