sipeed/NanoKVM #298: boot button ? What boot button...

adolfotregosa, 2025-02-05:
https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/user_guide.html

"""The main interface provides an OLED sleep function to prevent screen burn-in; a short press of the BOOT button can turn the OLED off or on."""

The nanokvm cube does not have a boot button or I'm blind. What am I missing ? Correct the user manual guys.

adolfotregosa, 2025-02-06:
Yeah, I see a hole, not a button, but yeah...

closed 2025-02-06 by adolfotregosa

vladyst, 2025-06-16:
I think it may be useful to route the "boot button" to one of the spare GPIO pins.
The web setup interface for changing the password apparently require to press the button. Accessing boot button is difficult as the nanoKVM is located behind other equipment.

jauling, 2026-09-05:
That's kind of crazy that the only way to wake the OLED screen is to press one of the most difficult buttons there is. Why not wake via RESET or POWER buttons?