sipeed/NanoKVM #300: KVM pings network gateway constantly

apalrd, 2025-02-06:
It's unusual for a device to send a ping every 2sec continuously. 

It's coming from chack_net_state() (https://github.com/sipeed/NanoKVM/blob/97a9b376c79611ca1cceb8f6df282670bb597379/support/kvm_system/main/lib/system_state/system_state.cpp#L146)

This is a terrible way to know if the network is 'up', and doesn't consider link-local networks which may not have a gateway, or IPv6.

alexisfrjp, 2025-02-17:
It's also a terrible way to spot a bad practice without giving at least one solution.

apalrd, 2025-02-17:
the kernel already provides network state, it's stupid to implement this again in an app

FlorianHeigl, 2025-03-02:
@alexisfrjp blameshifting, noone stops anyone who would like to fix this or design a product to first buy a few books on embedded development. we're here supporting it but if someone reports an issue *for free* for a product that is also developed *for free* the bargain is kept.

gshpychka, 2025-03-02:
Was this solved by https://github.com/sipeed/NanoKVM/pull/307?

Z2Z-GuGu, 2025-03-12:
On the latest 2.2.2 application, you can stop pinging the gateway by using the command `touch /etc/kvm/stop_ping && /etc/init.d/S95nanokvm restart`

> the kernel already provides network state, it's stupid to implement this again in an app

Additionally, I would like to point out that although the kernel provides network status, the status retrieved by the kernel is often incorrect due to the design of the internal PHY in the SG2002 chip. Using ping is a workaround. If you don't need to monitor the Ethernet status, you can disable the ping operation using the method above after updating the software. However, by default, I have kept it enabled, as having an unstable ETH status is not acceptable for a product.

Z2Z-GuGu, 2025-03-12:
> Was this solved by [#307](https://github.com/sipeed/NanoKVM/pull/307)?

I have tested that method, but unfortunately, it still doesn’t work. The status retrieved that way is also frequently incorrect.

closed 2025-03-12 by Z2Z-GuGu

apalrd, 2025-03-15:
I think closing this as completed is incorrect. It's still an issue, it's just caused by a hardware bug, so you are ignoring it