sipeed/NanoKVM #303: NanoKVM in combination with KVM switch

Michal-Szczepaniak, 2025-02-06:
Hi, since this is KVM i thought about buying KVM switch and using it with it, as you would do to access more devices, and sadly NanoKVM doesn't play nicely with it.

Two issues I've noted:

1. When switching the input, the encoding gets mangled, i managed to fix it if i switch to something else and quickly switch back. I don't know how it behaves on other switches because i have just one and i did not have any response from others who used it with kvm switch, but since its kvm i imagine its use in conjunction  with kvm switch is in mind. Here's video of the encoding bug

https://github.com/user-attachments/assets/59defa07-ab31-4984-8995-a907dc3c2454

2. The nanokvm's keyboard and mouse is not recognized by kvm, when plugged into kvm switch console input, the kvm switch does not recognize the nanokvm as mouse/keyboard, could be because the nanokvm's usb has multiple devices and it can't be resolved

azbesthu, 2025-02-14:
1. Is that h.264 video mode? I noticed also some distortion, even with some high detail background images. My experience is that h.264 has lower latency but more dissorted. MJPEG has much clearer picture, but can be slow.
2. What kind of KVM Switch do you have? Some KVMs only work with classic keyboards. I have good experience with XH-HK4401 #188 , Sometimes some process crashes in NanoKVM after switching inputs and it needs to be restarted, but otherwise it is usable.

Michal-Szczepaniak, 2025-02-14:
Mine is uniclass RL1086, i will try mjpeg but latency on h264 is already atrocious lol

azbesthu, 2025-02-14:
Try to connect to the usb port with "hub" writing on the other side of the switch.  The Uniclass RD-1086 seems to support hotkeys, so most likely the "console" usb port does something with the usb data flow to filter out those hotkey peresses. 
Maybe the "hub" port will work but you won't be able to use hotkeys for switching inputs. This is why I used serial controllable switch in my case.

Michal-Szczepaniak, 2025-02-14:
Yeah I know thats what i did and it works, but its workaround because i cannot switch input, i have to walk whole 5 meters to switch input, that's unacceptable

TBH for switching i might just hook up esp32 to click the buttons

azbesthu, 2025-02-14:
Maybe you need some serial/gpio command to esp32 keyboard emulator. 
When I searched for usable kvm switches I found some warning on some devices that control usb port is not even compatible with all real keyboards.

Michal-Szczepaniak, 2025-02-14:
nah i'm not gonna use it to emulate keyboard, im gonna hook it up to the front panel buttons and make it press them :P The key combos kinda suck cause you can't just switch to any port you want

Michal-Szczepaniak, 2025-02-15:
Update on that, mjpeg did not fix the bugging out issue

Reletiv, 2025-02-17:
Did you get it to work? Keyboard mouse also don't work on my HP kvm switch

Michal-Szczepaniak, 2025-02-17:
I mean with usb hub thingy yes it does pass through the keyboard/mouse/whateverelse, but i didn't get keyboard hotkeys to work neither does it work if i connect it to console input and it still bugs out when i switch input

but I added esp32 to switch input for me and combined with switching between inputs quickly to fix the broken encoding it somewhat works but still both things need to be fixed

Michal-Szczepaniak, 2025-06-06:
Update: with hid only mode it also doesn't work.