sipeed/NanoKVM #308: How to turn on the LED screen?

alexisfrjp, 2025-02-10:
When the NanoKVM boots up, the LED screen is on but after few seconds/minutes, it turns off.

How to turn it on again to read the IP address?

Z2Z-GuGu, 2025-02-17:
We have described this issue in the wiki: https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/faq.html#OLED-Not-Lighting-Up:1.6.1: Version 2.1.4 added an OLED sleep function; pressing the BOOT button can temporarily turn on the OLED. If you are using the NanoKVM-PCIe, the BOOT button is easy to find, located next to the HDMI interface, marked with 'BOOT.' If you are using the NanoKVM-Full/Lite, the BOOT button is on the right side of the USB interface; the Full version typically has a small hole, where you can press the switch inside with the included pin (using leverage). 
If you want to frequently check the information on the OLED, turning off the OLED sleep function in the settings may be a better option

closed 2025-02-17 by Z2Z-GuGu

alexisfrjp, 2025-02-17:
Perfect thank you!