sipeed/NanoKVM #309: 1.3.0 firmware completely broken

songokussm, 2025-02-13:
No video in 264, the default video mode. 
MJPEG shows 1 frame per second.
HUD doesn't work at all.
Terminal doesn't load.
Flashed succeeded.
Reflash succussed.

This is for the NanoKVM full.

Any ideas?

coolgoh98, 2025-02-17:
I have the same issue after updating. Any way to downgrade to previous 1.2.5 version. 
The instruction is not very clear how to do it. I have NanoKVM full

azbesthu, 2025-02-17:
I guess you can always download system images from the projects github releases page
https://github.com/sipeed/NanoKVM/releases
And you can put the microSD card from the device to a card reader/writer dongle and write older or the new image with Etcher utility (or any other image writer tool if you have other system than windows)
https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html

songokussm, 2025-02-17:
none of my fulls have a button.
flashing with etcher or rufus does not resolve.
using a different sdcard does not resolve.
chrome and firefox both have these issues.
I have two others running 1.2 with no issues.

Jonher937, 2025-02-21:
1.2.0 image is available here: https://github.com/sipeed/NanoKVM/releases/tag/v1.2.0
You need to unpack from xz soyou get a .img file before you flash it.

I was looking for the reset button myself, there's a hole but I didn't feel a reset button, after disassembling one I realised the reset button is flat (down) so you have to tilt the reset key so that it points down.
The tool should go almost the whole way in, then push the tool upwards so that the pointy part of the tool (inside the hole) pushes downwards. The button is quite "clicky" so you'll hear when you're in the right place.

Press and hole the button while connecting the USB-A end to a computer to access it without disassembling it.