sipeed/NanoKVM #311: [2.1.6] /etc/resolv.conf still lists Google and Chine DNS resolvers

axeleroy, 2025-02-14:
Hello,

The changelog for version 2.1.6 lists the following change

> perf: removed unnecessary modifications to DNS configuration

I understand it is in relation to [this answer](https://github.com/sipeed/NanoKVM/issues/301#issuecomment-2639491795) where you stated you would stop modifying the default DNS resolvers configuration.

Yet when I check the content of `/etc/resolv.conf`

```bash
~ # cat /etc/resolv.conf 
nameserver 192.168.0.1
nameserver 8.8.4.4
nameserver 8.8.8.8
nameserver 114.114.114.114
nameserver 119.29.29.29
nameserver 223.5.5.5
```

we can see 2 Google resolvers and 3 chinese (114dns,  dnspod and alidns)

Also, it does not follow the DNS resolver set in DHCP. In my case, it's `192.168.0.113` yet it calls `192.168.0.1`

```bash
nslookup google.com
Server:         192.168.0.1
Address:        192.168.0.1:53

Z2Z-GuGu, 2025-02-14:
Please modify the DNS you need in /boot/resolv.conf. After rebooting, /etc/resolv.conf will be copied from /boot/resolv.conf. However, version 2.1.6 will not make any changes to /boot/resolv.conf and /etc/resolv.conf

Z2Z-GuGu, 2025-02-14:
Due to the existence of this mechanism in the original SG2002 SDK (/etc/init.d/S02config), it has not been removed in this update for compatibility reasons.
In previous versions, to ensure that all users could access our updates, the program would first check if /etc/resolv.conf contained DNS entries like 223.5.5.5. If not, it would forcibly write to and overwrite both /boot/resolv.conf and /etc/resolv.conf. However, in version 2.1.6, the program will not overwrite these files at startup. Users are advised to configure a permanent DNS in /boot/resolv.conf and a temporary DNS in /etc/resolv.conf

axeleroy, 2025-02-14:
Good to know, maybe that should be documented somewhere. Anyway, setting my own DNS resolver in `/boot/resolv.conf` and rebooting does work as described

```bash
~ # cat /etc/resolv.conf 
nameserver 192.168.0.113

~ # nslookup -debug upu.samsungelectronics.com
Server:         192.168.0.113
Address:        192.168.0.113:53

Query #0 completed in 10ms:
authoritative answer:
Name:   upu.samsungelectronics.com
Address: 0.0.0.0

Query #1 completed in 14ms:
authoritative answer:
Name:   upu.samsungelectronics.com
Address: ::

~ # nslookup google.com
Server:         192.168.0.113
Address:        192.168.0.113:53

Non-authoritative answer:
Name:   google.com
Address: 2a00:1450:4007:81a::200e

Non-authoritative answer:
Name:   google.com
Address: 142.250.201.174
```

Z2Z-GuGu, 2025-02-17:
> maybe that should be documented somewhere

This is indeed our oversight, and I will add it to the wiki documentation as soon as possible. Thank you very much for your reminder.

closed 2025-02-17 by Z2Z-GuGu