sipeed/NanoKVM #314: Constantly being asked to change password

ajq11, 2025-02-15:
I have re-imaged the device many times now due to it becoming unaccessable throwing the ERR_CONNECTION_REFUSED error after a period of time working normally. Now, after the re-image the device constantly asks for the password to be reset and even after resetting the password, keeps popping up and showing no image.

Image version: v1.3.0
App version: 2.1.6

[image]

li20034, 2025-02-15:
Yeah, I noticed this as well. Seems to be a slight typo in the password updated check. See https://github.com/sipeed/NanoKVM/pull/312

Infamousbugg, 2025-02-16:
I had the same issue. I rebooted the NanoKVM after changing the password and got no prompt after that.

ajq11, 2025-02-16:
I have had to reboot it several times and run the update command several times due to the ERR_CONNECTION_REFUSED, which i also don't understand why it keeps happening.

Infamousbugg, 2025-02-16:
> I have had to reboot it several times and run the update command several times due to the ERR_CONNECTION_REFUSED, which i also don't understand why it keeps happening.

Did you give it the same password? I did use a different password after updating. I did see the same change password spam on both of my NanoKVM's (Full version + PCIe), a reboot fixed that issue on both of them.

ajq11, 2025-02-16:
> > I have had to reboot it several times and run the update command several times due to the ERR_CONNECTION_REFUSED, which i also don't understand why it keeps happening.
> 
> Did you give it the same password? I did use a different password after updating. I did see the same change password spam on both of my NanoKVM's (Full version + PCIe), a reboot fixed that issue on both of them.
> 
> Maybe try re-flashing using a different memory card if you continue to have issues. The only time I get a connection refused is when Firefox tries to connect over https.

Yeah, I change the password each time. 

Also, I did replace the memory card with a reputable one. Mine has the connection refuse with or without https. When I SSH in and reboot it, it seems to work for a random amount of time (usually within the same day).

li20034, 2025-02-16:
>  I did see the same change password spam on both of my NanoKVM's (Full version + PCIe), a reboot fixed that issue on both of them.

Did you actually change the password from the default? And are you on 2.1.6? Because, 2.1.6 has a bug where it actually will only prompt you to change the password, if your password is **not** the default of "admin" instead of if it is the default (like it's supposed to).

ajq11, 2025-02-16:
> Did you actually change the password from the default? And are you on 2.1.6? Because, 2.1.6 has a bug where it actually will only prompt you to change the password, if your password is **not** the [...]

Yes and yes. So, because I am usuing a differnt password, it is giving the error? lol.

Infamousbugg, 2025-02-16:
> Did you actually change the password from the default? And are you on 2.1.6? Because, 2.1.6 has a bug where it actually will only prompt you to change the password, if your password is **not** the default of "admin" instead of if it is the default (like it's supposed to).

After the system restarted following the update, I used the password change prompt to set a new password, using my own personal password. On my first device, I originally tried to give it the same password I had used when setting it up (which was also different than default). It took the password, but when the change password prompt came back I gave it a totally new password. On my second device I just gave it a new password from the get go, but also saw the same change password spam.

li20034, 2025-02-16:
> So, because I am usuing a differnt password, it is giving the error? lol.

Yes.

wj-xiao, 2025-02-17:
We're sorry for the mistake. Thanks to @li20034  for fixing it!

wj-xiao, 2025-02-18:
Fixed in version [2.2.0](https://github.com/sipeed/NanoKVM/releases/tag/2.2.0).

closed 2025-02-18 by wj-xiao