sipeed/NanoKVM #321: NanoKVM Lite - GPIO for "power" relay?

andriej, 2025-02-17:
I'm quite satisfied with what I've managed to do with NanoKVM already.
What I'm missing is possibility to turn on the machine remotely in case of failure. It's not an ATX anyway so I've soldered two cables directly to the power-switch.

Now I need to plan wiring it to NanoKVM Lite.
1. Which GPIO I should use to have possibility of using 'native' power button in UI?
2, Which relay I should order to have it triggered properly? Activated with LOW or HIGH signal with 3.3V?

Regarding to schematics I assume it's GPIO A23?

PatrickHuetter, 2025-02-18:
I'm also interested in this. Currently i'm using a 4-channel relay board with an ESP01 with tasmota on it to toggle power for 4 mini PCs. I would also rebuild this to be able to power toggle via nanoKVM Lite.

andriej, 2025-02-18:
I've ordered relay with NC/NO so no matter the trigger it should work. I hope NanoKVM does not trigger the gpio on reboot as it would shutdown the machine ;-)

staberas, 2025-02-18:
`
#!/bin/sh

# GPIO paths (adjust these paths as per your LicheeRV configuration)
GPIO_REBOOT="/sys/class/gpio/gpio505/value"
GPIO_POWER="/sys/class/gpio/gpio503/value"

# UART device (replace with the correct UART interface for LicheeRV Nano)
UART_DEVICE="/dev/ttyS1"
BAUD_RATE="115200"

# Initialize UART
stty -F $UART_DEVICE $BAUD_RATE cs8 -cstopb -parenb

# State tracking
LAST_REBOOT=0
LAST_POWER=0

# Monitor GPIOs and send events over UART
while true; do
    CURRENT_REBOOT=$(cat $GPIO_REBOOT)
    CURRENT_POWER=$(cat $GPIO_POWER)

    # Detect reboot button state change
    if [ "$CURRENT_REBOOT" -ne "$LAST_REBOOT" ]; then
        echo "event=reboot state=$CURRENT_REBOOT duration=0.0" > $UART_DEVICE
        LAST_REBOOT=$CURRENT_REBOOT
    fi

    # Detect power button state change
    if [ "$CURRENT_POWER" -ne "$LAST_POWER" ]; then
        # Long press handling
        if [ "$CURRENT_POWER" -eq 1 ]; then
            POWER_PRESS_START=$(date +%s)
        else
            POWER_PRESS_END=$(date +%s)
            PRESS_DURATION=$((POWER_PRESS_END - POWER_PRESS_START))
            echo "event=power state=0 duration=$PRESS_DURATION" > $UART_DEVICE
        fi
        LAST_POWER=$CURRENT_POWER
    fi

    sleep 0.1
done
`

here a script i made for my lite 
_tldr_: it sends over uart to an esp32 the poweron/off gpio state and then transmit it over ESP-NOW to another inside my pc.

Explanation of the Shell Script
**GPIO Monitoring:**

The script monitors the states of the GPIOs /sys/class/gpio/gpio505/value (reboot) and /sys/class/gpio/gpio503/value (power).

**UART Communication:**

Sends messages over the specified UART device (/dev/ttyS1).
Messages follow the format:
php-template
Copy
Edit
event=<event_name> state=<state_value> duration=<duration_value>
Example for a short press: event=power state=0 duration=1
Example for a reboot: event=reboot state=1 duration=0.0
Press Duration Calculation:

For the power button, the script calculates the duration between press and release (state=1 to state=0).
**Non-Blocking:**

The script runs in an infinite loop but does not block the system. It checks GPIOs every 0.1 seconds.

andriej, 2025-03-03:
I've received relay, connected it to A23 and it just works with default UI power button feature.

closed 2025-03-03 by andriej

fvillena, 2025-04-27:
> I've received relay, connected it to A23 and it just works with default UI power button feature.

3.3v or 5v relay?

andriej, 2025-04-27:
3.3V, from Aliexpress (like this one):

[image]