sipeed/NanoKVM #332: Can't boot to .iso inside /data folder after update.

mcury1985, 2025-02-19:
After updating to 1.4.0 and app version 2.2.0, can't boot to .iso anymore inside /data folder.

Mounted correctly, it shows up in nanoKVM as mounted.
In BIOS, it shows the drive but not the partition.

ISO I'm using can be found here: [https://endeavouros.com/](https://endeavouros.com/)

If you need more details to help, please let me know.
Thanks.

Z2Z-GuGu, 2025-02-20:
In the recent app update, we changed the image mounting mode to CD-ROM to be compatible with some very old BIOS. However, on some new motherboards, this causes the BIOS to be unable to detect the image. We plan to implement a switch to toggle this mode and will update it as soon as possible.

mcury1985, 2025-02-20:
Thanks for the feedback confirming the issue, this new version is awesome, really enjoying it

wj-xiao, 2025-02-21:
Fixed in [2.2.1](https://github.com/sipeed/NanoKVM/releases/tag/2.2.1)

mcury1985, 2025-02-21:
> Fixed in [2.2.1](https://github.com/sipeed/NanoKVM/releases/tag/2.2.1)

just updated to 2.2.1, and I can confirm, it is fixed.
Really thanks !

[image]

closed 2025-02-21 by mcury1985

jonny190, 2025-03-04:
I've had to roll back to [2.2.0](https://github.com/sipeed/NanoKVM/releases/tag/2.2.0) to get multipul systems to boot succsessfully or they get stuck at the bios when an ISO or Disk as mounted

alexislefebvre, 2025-10-05:
> In the recent app update, we changed the image mounting mode to CD-ROM to be compatible with some very old BIOS. However, on some new motherboards, this causes the BIOS to be unable to detect the image. We plan to implement a switch to toggle this mode and will update it as soon as possible.

When the CD-ROM mode is disabled, does it use another mode, like USB or HDD? I couldn't find any information about this on this repo or the wiki.