sipeed/NanoKVM #333: Recent update may have bricked my KVM

insergence, 2025-02-19:
I am not sure how to proceed but my issues stemed from the recent updates. Initially the kvm HID would stop working or the video wont not appear. Eventually the problem became worse especially with the latest updates. I can no longer access the WebUI. Hitting reset does not work either. Not sure how to troubleshoot this without an interface.

wj-xiao, 2025-02-20:
Sorry to hear that.
Can you access your device via SSH? Or unplug it from power and reconnect?

FlyingHelix, 2025-02-20:
I also do not have hdmi and keyboard only works intermittently. it started with the app update, where i could not login anymore, it always said something about an error with the password, but not that it was wrong. i then reflashed the microsd card with the latest version, thats when the HDMI stopped working. I can open the site, although sometimes i only get a complete white screen on Chrome on MacOS, but i was able to set a new passwort, change settings, login to tailscale, etc, but no video. i will try with an earlier image later today and post an udate if that works

heshamkhan1, 2025-02-20:
Mine did exactly the same, I sshed into mine and ran these, it fixed it for me:
```
cd /etc/kvm/
python update-nanokvm.py
```

FlyingHelix, 2025-02-21:
> Mine did exactly the same, I sshed into mine and ran these, it fixed it for me:
> 
> ```
> cd /etc/kvm/
> python update-nanokvm.py
> ```

that did it for me, thanks!

insergence, 2025-02-21:
How do you ssh into this? There are no troubleshooting guides for this.

insergence, 2025-02-21:
That did the trick! Thanks!

closed 2025-02-21 by insergence