sipeed/NanoKVM #339: Issue with 3440 x 1440

pgammel, 2025-02-20:
I am using a display port splitter to split my screen between my external monitor and Sipeed Nano KVM. There is no option to select 3440 x 1440, and the display on the nano KVM shows back when I connect it via active DP to HDMI adapter. I've tried several adapter combos, and I still can't get it to work properly.

ChokunPlayZ, 2025-02-21:
1920x1080 is the hard limit, the capture chip just can't go higher than this (I might be wrong).  
from my research you can make it work, you'll need a "Downscaler".  
I found a few devices out there, some have built in spliter and you can pick which output you want to scale down to 1080p.

pgammel, 2025-02-21:
Would you mind posting an example? I’m having a hard time finding
something. I’ve done several searches.

ChokunPlayZ, 2025-02-22:
[image]
here's the one I found, it has a built in spliter, can scale down 4K to 1080p.

pgammel, 2025-02-25:
> [image] here's the one I found, it has a built in spliter, can scale down 4K to 1080p.

I've tried that, but it does not output in 3440x1440 for my monitor. I tried several HDMI splitters (I can provide links if necessary) and none of them were able to output to monitors native resolution.

Here's a diagram of what I'm trying to do if it helps-

[image]

ChokunPlayZ, 2025-02-25:
have you tried this?

DP spliter -> DP to HDMI -> HDMI Scaler -> NanoKVM
     -> Monitor

the main output of the DP Spliter goes to your monitor, the secondary goes to the scaler.

pgammel, 2025-02-25:
Yes, but only on the DP splitter. I will order the one Wendell shows in his video, but I'm pretty sure it won't output 3440x1440 correctly to my monitor.

pgammel, 2025-02-25:
I just updated to 2.2.1 and it fixed the issue!

saucedge, 2025-03-17:
I'm considering going this route... I'm currently using an unused GPU HDMI port to feed the KVM, but that has me duplicating my main display to the KVM in windows which works just fine... until I want to fire up a game.  I assume Windows is handling downscaling to the KVM while any game just locks to 1080p when it sees the duplicate screens.  Your solution makes the KVM pretty much invisible in configuration, right?  Is there any significant latency or anything you are aware of?

pgammel, 2025-03-17:
I got the nano to show the 3440 x 1440 resolution, but after proper testing
I could not control it and found that the screen had locked up.

jpluimers, 2025-03-28:
> [image] here's the one I found, it has a built in spliter, can scale down 4K to 1080p.

It is this product (to check your local Amazon change the toplevel domain of the Amazon links):

- https://www.easycoolav.com/products/4k60-hdmi-splitter-1x2-dolby-vision-hdr-scaler-edid-setting-sp12h2
- https://www.amazon.com/-/en/Splitter-HDCP2-2-18Gbps-Vision-Monitors/dp/B083JVPXDY
- https://www.amazon.de/-/en/Splitter-HDCP2-2-18Gbps-Vision-Monitors/dp/B083JVPXDY

superphreak4, 2025-03-29:
Will these downscale 1440p 144hz? The descriptions don't mention it, or say "up to 4k"
Or maybe the updated SP12H21?

bilogic, 2025-04-04:
> I used this Monoprice DP Splitter https://www.amazon.com/dp/B074KTC3L6?ref=ppx_yo2ov_dt_b_fed_asin_title (Monoprice 1x2 DisplayPort Splitter, 4K@60Hz, 21.6Gbps)

Were there others that worked? This requires it's own power supply, I was hoping there is a DP that doesn't need external power