sipeed/NanoKVM #342: Unable to use UART1

KLUSEK, 2025-02-21:
Hi eveyrone
I just got my Nano and for a project I planned to used I wanted to make use of UART1 to communicate.

So, for testing purposes I used one of those USB to TTL converters and my laptop.

Following below pinout I've connected to UART1 RX, TX and Ground, double checking that I'm connecting it in TX<->RX manner.
[image]

But nothing really happens. 
What I tried:
- WebUI and /dev/ttyS1 and /dev/ttyS2, but I though maybe I don't know how to use picocom, so I logged in into shell and tried to send some data to my laptop using python and shell like that:
```

ser = serial.Serial('/dev/ttyS1', 115200)
ser.write(b'Hello')
ser.close()
```
```

stty -F /dev/ttyS1 115200 cs8 -cstopb -parenb
echo "Test" > /dev/ttyS1
```

But i remained silent. 
Then I tried to connect to UART0, which supposed to be system log printout and it worked right away.
Then I connected to UART2 and it started to work also.

At the final attempt I connected osciloscope to UART1 TX pin, but could not registed any signals.

Do you know why my UART1 does not working properly?
Or how to debug that problem?

azbesthu, 2025-02-23:
Did you connected pins correctly? I just asking, because S1 and S2 are not in the same order. 
I have two nanoKVM lite. One of them configured to use S1 and the other one uses S2 with python scripts. 
They are working fine for me (1.3.0 base image + updated application). 
Which image version do you have?

KLUSEK, 2025-02-23:
😶‍🌫️

That's a good catch! I can't remember when I get pinout image exactly, but this is interesting.
I'm abroad now, but next week I will confirm if that was it.

Thanks again!

wtarreau, 2025-03-02:
Indeed! The EMMC pins are also referred to as SPI1 on one and SPI4 on the other one!

KLUSEK, 2025-03-02:
Confirmed. Pinmap was wrong. Now it's working just fine :)

Thanks for help!

closed 2025-03-02 by KLUSEK