sipeed/NanoKVM #346: 1.4.0 Firmware issues

yodaphone, 2025-02-23:
I upgraded to the 1.4.0 firmware and the "Check for Updates" section says "get version failed" error

[image]

this the about NanoKVM

[image]

The image version says its "2025-02-17-19-08-3649fe.img" & not 1.4.0

When I try to use the mouse in the browser, it gets kickout of the window and I'm unable to interact with it

superphreak4, 2025-02-23:
The image version is the same for me. I questioned it myself, but it's correct. 
I would try a manual update of the applications 
https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/faq.html

yodaphone, 2025-02-25:
i noticed something else. the resolv.conf has 192.168.0.1 as the 1st IP but my IP range is something else. My DNS is 192.168.11.1 when i set that and save, it doesnt survive a reboot. it resets to 192.168.0.1

Jonher937, 2025-02-26:
It works fine for me, `Image Version: 1740524400` and frontend 2.2.1.
Check for updates reports: `You already have the latest version`.

For DNS settings to survive a reboot you want to edit `/boot/resolv.conf`

joshuagamboa, 2025-02-27:
I have the same issue (Image version showing as 2025-02-17-19-08-3649fe.img) and Check for Updates is failing. Tailscale is also not working. I tried to do: `python /etc/kvm/update-nanokvm.py` and just comes with the same error I've encountered back in firmware version 1.3.

So in 1.3, I needed to run this [alternate update](https://github.com/Bengt/NanoKVM-Updater) just to be able to update the frontend and finally use Tailscale. Does anyone know if this is still working? I don't want to run it this time at the risk of bricking the device since code hasn't been updated since 1.3.

yodaphone, 2025-02-27:
To check for updates, login to the terminal or ssh & modify the /etc/resolve.conf

the 1st line will default to 192.168.0.1
, if this isn’t your network change accordingly and also add 1.1.1.1 to it. Save. Give it a few seconds and then try again

yodaphone, 2025-02-27:
> I have the same issue (Image version showing as 2025-02-17-19-08-3649fe.img) and Check for Updates is failing. Tailscale is also not working. I tried to do: `python /etc/kvm/update-nanokvm.py` and [...]

You can't brick the device. Just toss in a fresh Microsoft

joshuagamboa, 2025-02-27:
> To check for updates, login to the terminal or ssh & modify the /etc/resolve.conf
> 
> the 1st line will default to 192.168.0.1
> , if this isn’t your network change accordingly and also add 1.1.1.1 to it. Save. Give it a few seconds and then try again
> […](#)

That totally worked out! Thanks.

joshuagamboa, 2025-02-27:
Now for Tailscale, I can't really make it work. Getting this error even when using `tailscale login` on ssh terminal.

Failed to connect to local Tailscale daemon for /localapi/v0/profiles/; not running? Error: dial unix /var/run/tailscale/tailscaled.sock: connect: no such file or directory

joshuagamboa, 2025-02-27:
Has anyone managed to get Tailscale working?

Dark-Existed, 2025-03-05:
Additionally, check the ```date``` and use ```date -s``` to set the correct time may help.