sipeed/NanoKVM #354: Can't update from 2.1.6

leodev12345, 2025-02-27:
So I recived my nanokvm full today and right off I noticed that I can't check for updates since in the update section it gives an error about version info so I tried to run the python script for manual updating and it also did not work. I had also previously set up a static ip adress with the provided instructions and set the ip to 192.168.1.200 and gateway to 192.168.1.1

## Problem 1. - wrong gateway settings

I accessed the nanokvm over ssh and right off i can see that there is no way to access the internet, pinging 8.8.8.8 gives an network unreachable error and i soon noticed that in the output of `ip r` the gateway route for eth0 is just wrong, its set to 192.168.1.0 for some weird reason even tho I never specified that adress and it seems to be hardcoded because when you reboot it just reverts back to it.

So to fix this issue I ran `ip route add default via 192.168.1.1 dev eth0` and that seemed to have fixed that isssue, I could now ping 8.8.8.8

## Problem 2. - wrong DNS settings

So with the gateway problem fixed I ran the script again and it still did not work, I also noticed that I can't still ping domains because the DNS config was wrong so I ran `cat /etc/resolve.conf` and all the nameservers were there, but there was something weird, there are actually 2 files: `resolv.conf` and `resolve.conf` and when you try to cat the resolv.conf it says that the file does not exist for some weird reason even tho it is clearly there? So to fix this i ran `vi /etc/resolv.conf` and entered `nameserver 8.8.8.8` and saved it and it worked right up, seems like all contents of resolve.conf should be in resolv.conf instead?

## Solution

When i added that route and fixed the DNS problem the update script downloaded the update and everything worked fine, rebooted but now all of the things I did are gone and there is again no internet or working DNS on the nanokvm, which is fine by me since I dont want this thing accessing anything outside my network anyway but I suppose this issue should be fixed.

Dark-Existed, 2025-03-05:
Maybe you should check ```/boot/resolv.conf```?

leodev12345, 2025-03-07:
Like I said if you do cat resolv.conf it says file not found, altho it does show in ls, maybe the file is corrupted or the whole image is corrupted or something?

leodev12345, 2025-03-10:
> Maybe you should check ```/boot/resolv.conf```?

that file doesent exist

Dark-Existed, 2025-03-10:
I am using ```v1.4.0``` image.

closed 2025-09-02 by leodev12345