sipeed/NanoKVM #357: Nanokvm is simply trash

delumerlino, 2025-02-28:
I am trying to make nanokvm work...but it is simply a white box with a screen.

- Using with vga to hdmi don't work on all devices (microserver G7 not work, G8 sometimes work)
- Cannot display during boot, so you cannot open bios settings
- When HID is connected, boot is delayed and when booted does not work as keyboard emulation. In some cases HID prevents system from booting.
- Soft keyboard is useless
- Paste keyboard is useless

Before trying wiring power buttons, I would like to know if I am the only one with this unusable box.
TY

techsupportkoeln, 2025-02-28:
yes, i have the same problem. Bought the pack of 5 and now also realized that I only see the display from the Windows login screen onwards. It's not possible to remote reinstall windows or access the bios.

delumerlino, 2025-02-28:
> yes, i have the same problem. Bought the pack of 5 and now also realized that I only see the display from the Windows login screen onwards on my Lenovo Laptop. It's not possible to remote reinstall windows or access the bios.

Pack of 5? Not so lucky....I try before end of return window...

Z2Z-GuGu, 2025-02-28:
I'm very sorry for the unpleasant experience. In a recent update, I attempted to fix the issue with VGA to HDMI. Please update to the latest app, which is currently version 2.2.1. Then, execute the following command in the web terminal: `echo 1 > hdmi_mode && reboot`. After that, wait for it to boot up, and it should display the image again shortly.

mbucc9000, 2025-02-28:
I'm having the same issue with my HP bios not recognizing HID. I've tried all the fixes mentioned in the few GitHub issues that have been posted. Do you know if this will be fixed in a future update? (HP Prodesk 600 G3).

ucki007, 2025-02-28:
Hi,

as the HID issue for BIOS / UEFI seems to be a main issue here, I'll crosslink two further topics regarding these issues:

#190 , #239

delumerlino, 2025-03-02:
> I'm very sorry for the unpleasant experience. In a recent update, I attempted to fix the issue with VGA to HDMI. Please update to the latest app, which is currently version 2.2.1. Then, execute the [...]

Nope, total black...

techsupportkoeln, 2025-03-03:
same here with the new firmware 1.4 and app version 2.2.1. :(

bkmartin, 2025-03-04:
this hdmi_mode fixed boot HDMI working for me... make sure you place the file in /etc/kvm

I'm using a VGA to HDMI adapter

techsupportkoeln, 2025-03-04:
i changed to the folder with cd /etc/kvm and made echo 1 > hdmi_mode && reboot
The KVM made then the reboot
I checked the hdmi_mode via cd /etc/kvm and cat hdmi_mode
There was a 1. 

But the problem is still there. I can't see the bios. The screen is only working in Windows.

techsupportkoeln, 2025-03-05:
The big question now is whether it is a hardware issue/limitation or can it be solved with a software update? Interestingly, JetKVM seems to have the same problem.
PiKVM does not have these problems.

delumerlino, 2025-03-05:
> i changed to the folder with cd /etc/kvm and made echo 1 > hdmi_mode && reboot The KVM made then the reboot I checked the hdmi_mode via cd /etc/kvm and cat hdmi_mode There was a 1.
> 
> But the problem is still there. I can't see the bios. The screen is only working in Windows.

You lucky! My console is Linux and it is total black...

Eisaichen, 2025-03-06:
I think this thing can only work with the VGA port, which is designed for server rack environment.
If I connect this to a HDMI port, the web server will keep crashing, making the whole thing unusable. The `echo 1 > /etc/kvm/hdmi_mode` will fix the crashing but still can only capture video after the system is fully booted, not working for UEFI, kind of useless.
And the overall stability is kind of poor, definitely does not qualify for an IT tool, I really wish I could get a refund for it.

delumerlino, 2025-03-06:
Luckily I have bought on Amazon.
I have started today the return process.

I will wait for a better kvm, or try pikvm.
Good luck to you all!

dragetd, 2025-03-06:
Since the issue is done, I'll allow myself to add a bit of a meta-comment:

@delumerlino
1. close the issue when you are done
2. use proper issue descriptions if you ever open an issue agian, that describe the actual problem
3. Write about what kind of diagnostics you found, what things you tried, what errors, logs, issues you concretely got

But most of all:
Watch your language. Do not blandly slander projects.

This kind of approach is highly unacceptable, neither commercial nor open source project. My assumption is that you do not have that much experience with open source or online communities in general, so please reflect on the way you communicate with others.

As for the maintainers of the repo, I think it is perfectly acceptable to close/delete/block such issues/persons without bad feelings. Save yourself the mental load and focus on the nice things.

hehongbo, 2025-04-09:
Log something just happened last night: I booted a LiveCD via Virtual Disk and went home from my office, leaving both the host and the NanoKVM running, but a few hours later, the virtual HID stopped working (HDMI streaming still works as I can tell from the flashing cursor) that I have to fix this by rebooting NanoKVM, and this lead to a crash of the LiveCD environment because the LiveCD media is disconnected when rebooting NanoKVM, leaving the machine a crashed state even Ctrl+Alt+Del can't fix...

...and my NanoKVM is a lite one so triggering a reboot via control-line levels won't be possible.

While NanoKVM is more compact compared to PiKVM, which I honestly like, it's not as stable. USB stops working from time to time, and the fact that the NanoKVM system will mount the entire exFAT `/data` as a USB storage when no ISO is selected, more switching would be involved when messing with images, that creates more opportunities for the whole thing to go wrong, especially in BIOS environments.