sipeed/NanoKVM #358: Home Assistant Support #Request

starlit-rocketship, 2025-02-28:
Hey,

Love the work of the project and love my 3x PCIe units.

I would really like to see someone create home assistant plugin for this. 
This may require some API to log in and have access to the power features?

My mian use case is automating power on of my PC when I push a smart button at my desk.
My PC is in my server rack in my garage, and booting it up remotely (other than loggin in to GUI) would be awesome.

Just putting it out there in case this project interests someone.

Z2Z-GuGu, 2025-02-28:
Hi, starlit-rocketship! 

I just thought of that too. Actually, I have similar ideas, but I've been really busy lately. I need to improve the existing NanoKVM-Lite/Full/PCIe and also develop NanoKVM-Pro (a more powerful version of NanoKVM), so this idea has been postponed indefinitely. Could you provide more information, such as the model of the desktop buttons? I might decide to research them one day in the future.

Snazzie, 2025-03-04:
Booting up remotely via HA should be easily done with Wake on lan Magic Packet? But i guess having authentication behind it could be useful.

starlit-rocketship, 2025-04-01:
That's really cool.
Thanks for putting the time into that.

I am not that savvy when it comes to coding but am happy to help test it.

Is there anything I can do to install and test it now or does it still need more work first?

vladitep, 2025-04-01:
probably a more universal solution (and probably more lightweight) would be to add mqtt support, then kvm can be added to any smart home systems, not just home assistant

laberet, 2025-04-24:
mqtt so it could be added to any home automation platform would be ideal

si458, 2026-09-26:
@Wouter0100 your integration doesn't work anymore with nanokvm 2.5.1
You need to update the python api from @puddly to the latest version a few days ago

puddly, 2026-09-26:
@si458 Can you open an issue in the respective projects? This doesn't feel like the best place to track it.

That being said, a patch release containing breaking API changes isn't great...

si458, 2026-09-26:
@puddly sure will do! No idea why I messaged here? Half asleep