sipeed/NanoKVM #362: USB connection constantly getting disabled and re-enabled

aptalca, 2025-03-02:
I hooked up the nanokvm full to a CWWK Intel N305 (also known as the purple Topton) board running Debian Bookworm and noticed that the server console was getting flooded with messages about USB getting disabled and re-enabled. Dmesg output confirmed it's the nanokvm usb connection.

[image]

 Here's the partial dmesg output on the server:

```
[79179.396027] usb usb3-port7: disabled by hub (EMI?), re-enabling...
[79179.396797] usb 3-7: USB disconnect, device number 104
[79179.940848] usb 3-7: new high-speed USB device number 105 using xhci_hcd
[79180.077985] usb 3-7: New USB device found, idVendor=3346, idProduct=1009, bcdDevice= 5.10
[79180.078004] usb 3-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[79180.078011] usb 3-7: Product: NanoKVM
[79180.078015] usb 3-7: Manufacturer: sipeed
[79180.078019] usb 3-7: SerialNumber: 0123456789ABCDEF
[79180.083040] input: sipeed NanoKVM as /devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.0/0003:3346:1009.028C/input/input660
[79180.225359] hid-generic 0003:3346:1009.028C: input,hidraw0: USB HID v1.01 Keyboard [sipeed NanoKVM] on usb-0000:00:14.0-7/input0
[79180.227382] input: sipeed NanoKVM as /devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.1/0003:3346:1009.028D/input/input661
[79180.227875] hid-generic 0003:3346:1009.028D: input,hidraw1: USB HID v1.01 Mouse [sipeed NanoKVM] on usb-0000:00:14.0-7/input1
[79180.229637] input: sipeed NanoKVM as /devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.2/0003:3346:1009.028E/input/input662
[79180.230057] hid-generic 0003:3346:1009.028E: input,hidraw2: USB HID v1.01 Mouse [sipeed NanoKVM] on usb-0000:00:14.0-7/input2
[79228.489026] usb usb3-port7: disabled by hub (EMI?), re-enabling...
[79228.489685] usb 3-7: USB disconnect, device number 105
[79229.048900] usb 3-7: new high-speed USB device number 106 using xhci_hcd
[79229.186012] usb 3-7: New USB device found, idVendor=3346, idProduct=1009, bcdDevice= 5.10
[79229.186031] usb 3-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[79229.186038] usb 3-7: Product: NanoKVM
[79229.186042] usb 3-7: Manufacturer: sipeed
[79229.186046] usb 3-7: SerialNumber: 0123456789ABCDEF
[79229.191217] input: sipeed NanoKVM as /devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.0/0003:3346:1009.028F/input/input663
[79229.333290] hid-generic 0003:3346:1009.028F: input,hidraw0: USB HID v1.01 Keyboard [sipeed NanoKVM] on usb-0000:00:14.0-7/input0
[79229.335441] input: sipeed NanoKVM as /devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.1/0003:3346:1009.0290/input/input664
[79229.335880] hid-generic 0003:3346:1009.0290: input,hidraw1: USB HID v1.01 Mouse [sipeed NanoKVM] on usb-0000:00:14.0-7/input1
[79229.337595] input: sipeed NanoKVM as /devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.2/0003:3346:1009.0291/input/input665
[79229.337953] hid-generic 0003:3346:1009.0291: input,hidraw2: USB HID v1.01 Mouse [sipeed NanoKVM] on usb-0000:00:14.0-7/input2
[79363.975027] usb usb3-port7: disabled by hub (EMI?), re-enabling...
[79363.975766] usb 3-7: USB disconnect, device number 106
[79364.452896] usb 3-7: new high-speed USB device number 107 using xhci_hcd
[79364.589974] usb 3-7: New USB device found, idVendor=3346, idProduct=1009, bcdDevice= 5.10
[79364.589993] usb 3-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[79364.590000] usb 3-7: Product: NanoKVM
[79364.590004] usb 3-7: Manufacturer: sipeed
[79364.590008] usb 3-7: SerialNumber: 0123456789ABCDEF
[79364.595390] input: sipeed NanoKVM as /devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.0/0003:3346:1009.0292/input/input666
[79364.745313] hid-generic 0003:3346:1009.0292: input,hidraw0: USB HID v1.01 Keyboard [sipeed NanoKVM] on usb-0000:00:14.0-7/input0
[79364.747452] input: sipeed NanoKVM as /devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.1/0003:3346:1009.0293/input/input667
[79364.747812] hid-generic 0003:3346:1009.0293: input,hidraw1: USB HID v1.01 Mouse [sipeed NanoKVM] on usb-0000:00:14.0-7/input1
[79364.749392] input: sipeed NanoKVM as /devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.2/0003:3346:1009.0294/input/input668
[79364.749799] hid-generic 0003:3346:1009.0294: input,hidraw2: USB HID v1.01 Mouse [sipeed NanoKVM] on usb-0000:00:14.0-7/input2
```

As you can see, every few minutes, the board receives gibberish from the nanokvm usb (interpreted as interference) and disables and re-enables it as a result.

I switched out the usb cable to a different one but same result.

I do have the nanokvm powered by a separate usb cable.

The bright side is, the re-enabling is successful most of the time, so the mouse and keyboard seem to work.

Any ideas?

Thanks

PS. I hooked up pikvm v3 to the same board and no issues with that

asychev, 2025-03-05:
Exactly the same issue on exactly the same board (and bunch of others too)

kariudo, 2025-03-18:
I see the same behavior on an N100 Beelink device running Arch.