sipeed/NanoKVM #366: tailscale 50 MB limit causes issues, needs to be raised to 75 MB

kinghrothgar, 2025-03-05:
I have all the latest versions installed, NanoKVM 2.2.1, Firmware 1.4.0, tailscale 1.80.3.  My connection to the Web App over tailscale kept breaking. With the memory limit not set, it eventually consumes too much memory and breaks.  With the memory limit set, it also was breaking. In both instances, the only way to fix it was to restart the NanoKVM.

After doing some debugging, I discovered that settings the GOMEMLIMIT to 50 MB causes tailscale to GC constantly. Even after doing `tailscale down`, `tailscaled` was using around +20% of the CPU. I did a go CPU profile, and it was constantly stuck in GC. 50 MB of vmem just isn't enough for the latest tailscales it appears. After tweaking it, I found that setting `/etc/kvm/GOMEMLIMIT` to 75 seemed to be the correct balance. I have been able to keep a Web App session open for over an hour and actual `tailscaled` memory usage  hovers around 40 MB.

wj-xiao, 2025-03-11:
Thank you for the suggestion. I've set the GOMEMLIMIT to 75MB in the latest version [2.2.2](https://github.com/sipeed/NanoKVM/releases/tag/2.2.2).

ps: If Tailscale is already running, please stop it and then start it again for the setting to take effect.

kinghrothgar, 2025-03-16:
@wj-xiao thank you! I tried to figure out where that was being set to do a PR but could :|

closed 2025-03-16 by kinghrothgar