sipeed/NanoKVM #370: Security vs usabilty, how it should be for tinker consumer devices. (please read before closing or deleting)

LeonStraathof, 2025-03-08:
The commotion about the security aspects of this product are almost all valid. The defence from the developer saying that usability is also important and sometimes goes against security best practice also is true in edge cases. The problem here is initially the developer deciding to set insecure defaults. This kind of device should always be setup first time in a lan environment where connectivity problems that the developer uses as defence do not exist. A good device would have some sort of first time setup wizzard that goes across the settings one by one that can open features that are less secure. (many routers have first start wizzards, so it is nothing new or difficult for consumers) That first setup wizzard preferrably should mention if there are security risks with some of the settings. You are afraid that users forget their password sure give them the option not to change it, but it has to complain several times that it is not recommended. You want to give them extra DNS servers, sure explain why they can help some users and then let them choose between the suggested or their own.

And last give them the option to run the wizzard again if they found their choices did not give the result they wanted. People should be warned about dangerous things in life but you cannot babysit them. This is a devices that has mondane users but also tinkerers that sometimes want to do things in the not recommended way and they should have that option. 

So in short this kind of device should always startup out of the box safe, and if users want to make it less safe that should be an option. And if they choose those unsafe options they should be warned preferrably with a short explenation why. But not stopped if they really want it.

So please look at all things again Apalrd mentioned they are valid points. And if you really think some users still want something that is less safe or not a wanted option by everyone put it in that first use wizzard.

PatrickHuetter, 2025-03-08:
@LeonStraathof mostly everything is already fixed. Dependencies were removed and everything was open sourced. What's missing (like some configuration options can be done through cli/ssh) And there is also a full open source firmware fork from @scpcom on ubuntu/debian base available. I'm already running it and it works fine. So you can just use basic tools like ansible to fully provision like you want (e.g. placing ssh authorized keys, configuring fail2ban, tailscale etc.).

wj-xiao, 2025-03-11:
Please refer to #301 .

LeonStraathof, 2025-04-15:
> [@LeonStraathof](https://github.com/LeonStraathof) mostly everything is already fixed. Dependencies were removed and everything was open sourced. What's missing (like some configuration options can [...]

It is moving in the right direction, and i do not think it is the biggest issue for myself and for you as you refer to using cli/ssh. But this device is sold not only to tinkerers but also to not tech savvy normal consumers. That's why everything should always start with safe defaults out of the box (not for you and me but for those who do not know what ssh is and how to do settings outside the web gui). Not every device type has the same attack potential, this type of device has very high potential. You know when looking at OS CVE's you will notice a lot have low threat scores because you need physical access to the device to exploit something. Problem is a KVM device like this gives bad actors that physical access to the machine you connect the KVM to but then remotely if it is not secure.