sipeed/NanoKVM #374: Video stream over reverse proxy

ajvcorreia, 2025-03-11:
I have a nano kvm sitting behind an NGINX Proxy Manager reverse proxy.
The web interface comes up and all options are available but not the video signal.
Is there a special configuration I should be using in NGINX to allow the video signal to pass through?

netandreus, 2025-03-18:
I don't know why, but when I opened the web ui in Chrome incognito mode I could see video output.

szsong, 2025-03-25:
Have you tried swiching video mode from H264 to MJPEG? That solved the problem for me. but I have another issue, the HID are not working when I connect through reverse proxy, but works normally if I use local IP address to access..

saucedge, 2025-03-27:
I'm having this issue

kuroiookami77, 2025-03-28:
The same problem - NanoKVM behind Nginx Proxy. When accessing from an external network, there is no video in H264, but if switched to MJPEG, it works fine.

dobrozhanskyi, 2025-04-01:
you need to configure TURN server for WebRTC,  you can see example in [doc](https://github.com/sipeed/NanoKVM/blob/main/server/README.md#configuration) 

```
# Address and authentication for custom TURN server
turn:
    turnAddr: example_addr
    turnUser: example_user
    turnCred: example_cred
```

ajvcorreia, 2025-04-06:
MJPEG worked! could swear I tried that but guess not....

faultoverload, 2025-04-08:
Have you enabled websocket support in NGINX proxy manager? I was having this issue and that setting resolved it.

ajvcorreia, 2025-04-08:
> Have you enabled websocket support in NGINX proxy manager? I was having this issue and that setting resolved it.

Yes, I had allready enabled that, but it was not enough, changing the video mode to MJPEG was what got it working for me.