sipeed/NanoKVM #376: No Display Output aft update to 2.2.2 (1.4.0)

th3cube, 2025-03-12:
mmf_add_vi_channel..
mmf_add_vi_channel..
mmf_add_vi_channel..
[2025-03-12 18:11:11] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:12] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:31] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:33] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:34] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:35] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:36] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:37] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:38] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:39] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:40] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:41] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:42] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:43] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:44] [error] [cgo.go:93] failed to read kvm image: -1

I tried serval things like reboot 
force update-nanokvm.py

setting manual resolution 
echo 1080 > /kvmapp/kvm/height etc.

nothing helps. The Software is very unstable. Product unusable...

Z2Z-GuGu, 2025-03-13:
Please provide more information to help troubleshoot the issue:
Execute `cat /etc/kvm/hdmi_version` and `cat /proc/cvitek/vi_dbg` separately in the web terminal.

chchia, 2025-03-13:
20250313 13:28:24.282 1111 E isp setNoiseProfile:3460 Noise profile get fail. Please check
********************************************************************************
cvi_bin_isp message
gerritId:      97347          commitId:      fc6ce647
md5:           d6db2297ddfd44e8252c1f3f888f47b2
sensorNum      1
sensorName0    6911

PQBIN message
gerritId:
              commitId:      fc6ce647
md5:           d6db2297ddfd44e8252c1f3f888f47b2
sensorNum      1
sensorName0    4653

author:        lxowalle       desc:          gc4653 30fps
createTime:    2024-03-26 13:41:34version:
tool Version:       v3.0.5.48           mode:
********************************************************************************
sensorName(0) mismatch, mwSns:6911 != pqBinSns:4653
[SAMPLE_COMM_ISP_Thread]-95: ISP Dev 0 running!
pstSize->u32Width = 1920
pstSize->u32Height = 1080
maix multi-media init ok
try release vio ok
try release venc ok
mmf_add_vi_channel..
2025/03/13 13:28:24 proto: http, port: 80 443
mmf_add_vi_channel..
mmf_add_vi_channel..
mmf_add_vi_channel..
[2025-03-13 13:28:36] [error] [cgo.go:64] failed to read kvm image: -1
[2025-03-13 13:28:37] [error] [cgo.go:64] failed to read kvm image: -1
[2025-03-13 13:28:38] [error] [cgo.go:64] failed to read kvm image: -1
[2025-03-13 13:28:39] [error] [cgo.go:64] failed to read kvm image: -1
[2025-03-13 13:28:40] [error] [cgo.go:64] failed to read kvm image: -1
[2025-03-13 13:28:41] [error] [cgo.go:64] failed to read kvm image: -1
[2025-03-13 13:28:42] [error] [cgo.go:64] failed to read kvm image: -1
[2025-03-13 13:28:43] [error] [cgo.go:64] failed to read kvm image: -1
[2025-03-13 13:28:44] [error] [cgo.go:64] failed to read kvm image: -1
[2025-03-13 13:28:45] [error] [cgo.go:64] failed to read kvm image: -1
[2025-03-13 13:28:46] [error] [cgo.go:64] failed to read kvm image: -1
[2025-03-13 13:28:47] [error] [cgo.go:64] failed to read kvm image: -1
[2025-03-13 13:28:48] [error] [cgo.go:64] failed to read kvm image: -1
[2025-03-13 13:28:49] [error] [cgo.go:64] failed to read kvm image: -1
[2025-03-13 13:28:50] [error] [cgo.go:64] failed to read kvm image: -1

```
# cat /etc/kvm/hdmi_version
c

# cat /proc/cvitek/vi_dbg
[VI BE_Dbg_Info]
VIPreBEDoneSts          :0x0            VIPreBEDmaIdleStatus    :0x0
[VI Post_Dbg_Info]
VIIspTopStatus          :0x0
[VI DMA_Dbg_Info]
VIWdma0ErrStatus        :0x0    VIWdma0IdleStatus       :0x0
VIWdma1ErrStatus        :0x0    VIWdma1IdleStatus       :0x0
VIRdmaErrStatus         :0x0    VIRdmaIdleStatus        :0x0
[VI ISP_PIPE_A FE_Dbg_Info]
VIPreFERawDbgSts        :0x0            VIPreFEDbgInfo          :0x0
[VI ISP_PIPE_A]
VIOutImgWidth           :1920
VIOutImgHeight          :1080
VIInImgWidth            :1920
VIInImgHeight           :1080
VIDevFPS                :   0
VIFPS                   :   0
VISofCh0Cnt             :   0
VIPreFECh0Cnt           :   0
VIPreBECh0Cnt           :   0
VIPostCnt               :   0
VIDropCnt               :   0
VIDumpCnt               :   0
[VI ISP_PIPE_A Csi_Dbg_Info]
VICsiIntStatus0         :0x0
VICsiIntStatus1         :0x0
VICsiCh0Dbg             :0x0
VICsiCh1Dbg             :0x0
VICsiOverFlowCnt        :   0
VICsiCh0WidthGTCnt      :   0
VICsiCh0WidthLSCnt      :   0
VICsiCh0HeightGTCnt     :   0
VICsiCh0HeightLSCnt     :   0
#
```

th3cube, 2025-03-13:
```bash
# cat /etc/kvm/hdmi_version
c
```

```bash
# cat /etc/kvm/hdmi_version
c
#
#
#
# ld
-sh: ld: not found
# cat /etc/kvm/hdmi_version
c
# cat /proc/cvitek/vi_dbg
[VI BE_Dbg_Info]
VIPreBEDoneSts		:0x0		VIPreBEDmaIdleStatus	:0x0
[VI Post_Dbg_Info]
VIIspTopStatus		:0x0
[VI DMA_Dbg_Info]
VIWdma0ErrStatus	:0x0	VIWdma0IdleStatus	:0x0
VIWdma1ErrStatus	:0x0	VIWdma1IdleStatus	:0x0
VIRdmaErrStatus		:0x0	VIRdmaIdleStatus	:0x0
[VI ISP_PIPE_A FE_Dbg_Info]
VIPreFERawDbgSts	:0x0		VIPreFEDbgInfo		:0x0
[VI ISP_PIPE_A]
VIOutImgWidth		:1920
VIOutImgHeight		:1080
VIInImgWidth		:1920
VIInImgHeight		:1080
VIDevFPS		:   0
VIFPS			:   0
VISofCh0Cnt		:   0
VIPreFECh0Cnt		:   0
VIPreBECh0Cnt		:   0
VIPostCnt		:   0
VIDropCnt		:   0
VIDumpCnt		:   0
[VI ISP_PIPE_A Csi_Dbg_Info]
VICsiIntStatus0		:0x0
VICsiIntStatus1		:0x0
VICsiCh0Dbg		:0x0
VICsiCh1Dbg		:0x0
VICsiOverFlowCnt	:   0
VICsiCh0WidthGTCnt	:   0
VICsiCh0WidthLSCnt	:   0
VICsiCh0HeightGTCnt	:   0
VICsiCh0HeightLSCnt	:   0
```

papou-github, 2025-03-15:
Take a look to my personal experience to solve the issue: [https://github.com/sipeed/NanoKVM/issues/179#issuecomment-2726383892](url)
Hope it will help

wlallemand, 2025-03-15:
Hello, I have the same issue, the device was working correctly with image 1.4.0 + software 2.2.1. I just updated to 2.2.2 and I can't get an image anymore.

```
# cat /etc/kvm/hdmi_version
ux
# cat /proc/cvitek/vi_dbg
[VI BE_Dbg_Info]
VIPreBEDoneSts		:0x0		VIPreBEDmaIdleStatus	:0x7
[VI Post_Dbg_Info]
VIIspTopStatus		:0x3ff
[VI DMA_Dbg_Info]
VIWdma0ErrStatus	:0x3000000	VIWdma0IdleStatus	:0xffffffff
VIWdma1ErrStatus	:0x3000000	VIWdma1IdleStatus	:0xffffffff
VIRdmaErrStatus		:0x3000000	VIRdmaIdleStatus	:0xffffffff
[VI ISP_PIPE_A FE_Dbg_Info]
VIPreFERawDbgSts	:0x3ff		VIPreFEDbgInfo		:0x3f
[VI ISP_PIPE_A]
VIOutImgWidth		: 800
VIOutImgHeight		: 600
VIInImgWidth		: 800
VIInImgHeight		: 600
VIDevFPS		:  60
VIFPS			:   0
VISofCh0Cnt		:19197
VIPreFECh0Cnt		:   0
VIPreBECh0Cnt		:   0
VIPostCnt		:   0
VIDropCnt		:   0
VIDumpCnt		:   0
[VI ISP_PIPE_A Csi_Dbg_Info]
VICsiIntStatus0		:0x3
VICsiIntStatus1		:0x0
VICsiCh0Dbg		:0x0
VICsiCh1Dbg		:0x0
VICsiOverFlowCnt	:   0
VICsiCh0WidthGTCnt	:  32
VICsiCh0WidthLSCnt	:   0
VICsiCh0HeightGTCnt	:   0
VICsiCh0HeightLSCnt	:   0
# 
```

ashleyw-gh, 2025-03-16:
I had the same issue of the UI failing after the upgrade to v2.2.2 (from 2.2.1).
In my case I had to manually do the update by connecting via SSH to the IP of the NanoKVM;
```
# cd /etc/kvm/
# python update-nanokvm.py
# reboot
```
the first time it appeared to complete but after the reboot the UI didn't come up, so I then repeated the process again via SSH;
```
# cd /etc/kvm/
# python update-nanokvm.py
# reboot
```
and then things worked as expected and the NanoKVM stuck on release 2.2.2.

wlallemand, 2025-03-16:
Unfortunately I tried that multiple time already, and this doesn't work more, the screen is still black on the webpage. I also tried multiple reboot and to unplug and re-plug the HDMI cable.  Only a downgrade make it work again.

geotekberlin, 2025-03-16:
Same thing here. After upgrading via web console from 2.1.5 to 2.2.2 the console screen is black while the web gui was still working and showed the correct version 2.2.2.

Reinstalling via update-nanokvm.py did not help, unfortunately it caused the /root/old backup directory of the previous version to be overwritten.

Manually extracting and installing the versions 2.2.1, 2.1.6 or 1.4.0 caused the web server to be completely unavailable, only ssh is still working. So now I am stuck with an unusable KVM that can't even be downgraded any more. Any suggestions?

Z2Z-GuGu, 2025-03-18:
It seems that the issue encountered by @chchia and @th3cube is due to VI not being properly initialized. Please execute the following steps in SSH to help me troubleshoot the problem:

```shell
# Enable the SSH switch on the webpage
# Use ssh root@xxx.xxx.xxx.xxx and the webpage password you set to enter SSH
# Use vi (or nano for image versions >= 1.4) to modify the value of logger->level in /etc/kvm/server.yaml to debug
/etc/init.d/S95nanokvm restart
```

Z2Z-GuGu, 2025-03-18:
I just reproduced this issue. You can use the following method: execute /etc/init.d/S95nanokvm restart, then change the resolution to 1920x1080 in the webpage, and the video will resume. This issue will be fixed in the next version.Thank you all for identifying the issue!

chchia, 2025-03-20:
``> It seems that the issue encountered by [@chchia](https://github.com/chchia) and [@th3cube](https://github.com/th3cube) is due to VI not being properly initialized. Please execute the following steps in SSH to help me troubleshoot the problem:
> 
> # Enable the SSH switch on the webpage
> # Use ssh root@xxx.xxx.xxx.xxx and the webpage password you set to enter SSH
> # Use vi (or nano for image versions >= 1.4) to modify the value of logger->level in /etc/kvm/server.yaml to debug
> /etc/init.d/S95nanokvm restart

``
**_you know how we get this log below? it was by running this exact command /etc/init.d/S95nanokvm restart_**

```
mmf_add_vi_channel..
mmf_add_vi_channel..
mmf_add_vi_channel..
[2025-03-12 18:11:11] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:12] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:31] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:33] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:34] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:35] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:36] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:37] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:38] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:39] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:40] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:41] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:42] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:43] [error] [cgo.go:93] failed to read kvm image: -1
[2025-03-12 18:11:44] [error] [cgo.go:93] failed to read kvm image: -1
```

wlallemand, 2025-03-23:
I just updated to 2.2.4 and the problem sill persist, after restarting S95nanokvm, error are looping this way:

```
[2025-03-23 21:22:51.748] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:51.780] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:51.814] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:51.848] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:51.881] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:51.916] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:51.947] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:51.984] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.014] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.048] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.080] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.114] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.148] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.180] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.214] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.247] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.281] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.314] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.347] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.381] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.414] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.447] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.481] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.514] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.548] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.582] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.614] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv] CSI abnormal: width too small
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.648] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.681] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.715] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.747] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.781] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:52.815] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[... 24 lines cut ...]
[2025-03-23 21:22:53.248] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:53.281] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:53.314] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:53.347] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:53.382] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:53.414] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:53.448] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:53.481] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:53.515] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:53.547] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:53.581] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-03-23 21:22:53.614] [error] [cgo.go:85] failed to read kvm image: -4

```

wlallemand, 2025-03-23:
I'm even more confused, I can't make it work again by downgrading to 2.2.1 anymore... I don't know if the upgrade changed something else than the /kvmapp/ directory that could be provoking this.

Z2Z-GuGu, 2025-03-24:
@wlallemand Perhaps I need more logs to diagnose the issue you're facing. Please follow these steps to open and view the logs:

1. Enable the SSH feature in the settings, and log in using `ssh root@192.168.xxx.xxx`. The password is the same as your web password.
2. Edit `/etc/kvm/server.yaml` and change `logger->level: info` to `debug`.
3. Restart with `/etc/init.d/S95nanokvm restart`.

wlallemand, 2025-04-01:
I  rolled back the SD with an 1.4.0 image and this was working again, however once the software updated to 2.2.5 I have a black screen instead of the HDMI output.

Here the debug output with 2.2.5:

[kvm.debug.log](https://github.com/user-attachments/files/19555470/kvm.debug.log)

I tried multiple things like changing the FPS, resolution, video mode, even reset the HDMI but does not seem to work more.

wlallemand, 2025-04-24:
Upgraded to 2.2.6, still broken. I can't get the display output at all.

Z2Z-GuGu, 2025-04-25:
@wlallemand Please execute `echo 0 > /etc/kvm/hdmi_mode` to see if it can be restored.

Z2Z-GuGu, 2025-04-25:
@Toasterme Please provide more information to help resolve the issue: execute `cat /etc/kvm/hdmi_version`, `cat /etc/kvm/hdmi_mode` and  `cat /proc/cvitek/vi_dbg` in the web terminal.

Toasterme, 2025-04-25:
> [@Toasterme](https://github.com/Toasterme) Please provide more information to help resolve the issue: execute `cat /etc/kvm/hdmi_version`, `cat /etc/kvm/hdmi_mode` and `cat /proc/cvitek/vi_dbg` in the web terminal.

thanks for replying, as i posted on another post, i solved it myself.  Even when a monitor the computer was not finishing its post.  As i played with the memory, it wasn't properly inserted.  So bottom line, make sure the post is passed and the computer is working before trying the kvm ;P

wlallemand, 2025-05-13:
I updated today to version 2.2.7, still not working for me. Last working version was 2.2.1 on the base 1.4.0 image.

wlallemand, 2025-05-13:
So I just tested with a new NanoKVM-PCIe board to be sure that the first wasn't faulty and I have the same problems, when using 2.2.1 I have an image, but once I upgrade this is a black screen.

```
# cat /etc/kvm/hdmi_version
ux
# cat /etc/kvm/hdmi_mode
2
# cat /proc/cvitek/vi_dbg
[VI BE_Dbg_Info]
VIPreBEDoneSts		:0x0		VIPreBEDmaIdleStatus	:0x7
[VI Post_Dbg_Info]
VIIspTopStatus		:0x3ff
[VI DMA_Dbg_Info]
VIWdma0ErrStatus	:0x3000000	VIWdma0IdleStatus	:0xffffffff
VIWdma1ErrStatus	:0x3000000	VIWdma1IdleStatus	:0xffffffff
VIRdmaErrStatus		:0x3000000	VIRdmaIdleStatus	:0xffffffff
[VI ISP_PIPE_A FE_Dbg_Info]
VIPreFERawDbgSts	:0x3ff		VIPreFEDbgInfo		:0x3f
[VI ISP_PIPE_A]
VIOutImgWidth		: 800
VIOutImgHeight		: 600
VIInImgWidth		: 800
VIInImgHeight		: 600
VIDevFPS		:  60
VIFPS			:   0
VISofCh0Cnt		:6485
VIPreFECh0Cnt		:   0
VIPreBECh0Cnt		:   0
VIPostCnt		:   0
VIDropCnt		:   0
VIDumpCnt		:   0
[VI ISP_PIPE_A Csi_Dbg_Info]
VICsiIntStatus0		:0x3
VICsiIntStatus1		:0x0
VICsiCh0Dbg		:0x0
VICsiCh1Dbg		:0x0
VICsiOverFlowCnt	:   0
VICsiCh0WidthGTCnt	:  32
VICsiCh0WidthLSCnt	:   0
VICsiCh0HeightGTCnt	:   0
VICsiCh0HeightLSCnt	:   0
# 
```
```
[...]
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:49.343] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:49.360] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:49.376] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:49.394] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:49.409] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:49.426] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:49.443] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:49.460] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:49.477] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:49.494] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:49.510] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:49.526] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[... 223 lines cut ...]
[2025-05-13 22:45:51.393] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:51.410] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:51.426] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:51.443] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:51.460] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:51.477] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:51.493] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:51.509] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:51.527] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:51.543] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:51.560] [error] [cgo.go:85] failed to read kvm image: -4
[kvmv]kvmv_read_img type = 1...
[2025-05-13 22:45:51.577] [error] [cgo.go:85] failed to read kvm image: -4
[...]
```

wlallemand, 2025-05-16:
@Z2Z-GuGu any update on this ? I can run any test needed.

inzi, 2025-06-04:
I'm having the same issue. 

```
# cat /etc/kvm/hdmi-version
cat: can't open '/etc/kvm/hdmi-version': No such file or directory
# cat /etc/kvm/hdmi_version
c
# cat /etc/kvm/hdmi_mode 
2
# cat /proc/cvitek_vi_dbg
cat: can't open '/proc/cvitek_vi_dbg': No such file or directory
# at /proc/cvitek/vi_dbg
/bin/sh: at: not found
# cat /proc/cvitek/vi_dbg
[VI BE_Dbg_Info]
VIPreBEDoneSts          :0x0            VIPreBEDmaIdleStatus    :0x0
[VI Post_Dbg_Info]
VIIspTopStatus          :0x0
[VI DMA_Dbg_Info]
VIWdma0ErrStatus        :0x0    VIWdma0IdleStatus       :0x0
VIWdma1ErrStatus        :0x0    VIWdma1IdleStatus       :0x0
VIRdmaErrStatus         :0x0    VIRdmaIdleStatus        :0x0
[VI ISP_PIPE_A FE_Dbg_Info]
VIPreFERawDbgSts        :0x0            VIPreFEDbgInfo          :0x0
[VI ISP_PIPE_A]
VIOutImgWidth           :1920
VIOutImgHeight          :1080
VIInImgWidth            :1920
VIInImgHeight           :1080
VIDevFPS                :   0
VIFPS                   :   0
VISofCh0Cnt             :   0
VIPreFECh0Cnt           :   0
VIPreBECh0Cnt           :   0
VIPostCnt               :   0
VIDropCnt               :   0
VIDumpCnt               :   0
[VI ISP_PIPE_A Csi_Dbg_Info]
VICsiIntStatus0         :0x0
VICsiIntStatus1         :0x0
VICsiCh0Dbg             :0x0
VICsiCh1Dbg             :0x0
VICsiOverFlowCnt        :   0
VICsiCh0WidthGTCnt      :   0
VICsiCh0WidthLSCnt      :   0
VICsiCh0HeightGTCnt     :   0
VICsiCh0HeightLSCnt     :   0
# 

```

inzi, 2025-06-04:
After some tinkering - I set it to automatic resolution, did the restart command, it scrolled an error, then via the web ui, I set the resolution manually. 

Video returned, and this was the output in ssh:

```
[2025-06-04 22:32:59.085] [error] [cgo.go:85] failed to read kvm image: -4
[2025-06-04 22:32:59.101] [error] [cgo.go:85] failed to read kvm image: -4
[2025-06-04 22:32:59.118] [error] [cgo.go:85] failed to read kvm image: -4
[2025-06-04 22:32:59.135] [error] [cgo.go:85] failed to read kvm image: -4
[2025-06-04 22:32:59.151] [error] [cgo.go:85] failed to read kvm image: -4
[2025-06-04 22:32:59.168] [error] [cgo.go:85] failed to read kvm image: -4
[2025-06-04 22:32:59.185] [error] [cgo.go:85] failed to read kvm image: -4
[2025-06-04 22:32:59.202] [error] [cgo.go:85] failed to read kvm image: -4
[2025-06-04 22:32:59.218] [error] [cgo.go:85] failed to read kvm image: -4
[2025-06-04 22:32:59.235] [error] [cgo.go:85] failed to read kvm image: -4
[2025-06-04 22:32:59.251] [error] [cgo.go:85] failed to read kvm image: -4
[2025-06-04 22:32:59.268] [error] [cgo.go:85] failed to read kvm image: -4
[2025-06-04 22:32:59.504] [error] [operation.go:16] open /dev/hidg0 failed: open /dev/hidg0: no such device or address
[2025-06-04 22:32:59.513] [error] [operation.go:21] open /dev/hidg1 failed: open /dev/hidg1: no such device or address
[2025-06-04 22:33:01.334] [error] [mouse.go:104] write to hid failed: invalid argument
[2025-06-04 22:33:01.414] [error] [mouse.go:104] write to hid failed: invalid argument
[2025-06-04 22:33:02.210] [error] [mouse.go:104] write to hid failed: invalid argument
[2025-06-04 22:33:02.314] [error] [mouse.go:104] write to hid failed: invalid argument
[2025-06-04 22:33:03.084] [error] [operation.go:66] write to hid failed: invalid argument
[2025-06-04 22:33:03.180] [error] [operation.go:66] write to hid failed: invalid argument
```