sipeed/NanoKVM #378: use of watchdog for NanoKVM-Server

prasaanth-slx, 2025-03-13:
It is seen that in release 2.2.2 [58d5ab2](https://github.com/sipeed/NanoKVM/commit/58d5ab2d37244b1e1a68b925a5c23c324c489ad3) (2025-03-11) feat: add watchdog for NanoKVM-Server is mentioned.  Can someone please explain how to configure it.

Is it for monitoring the connected host pc or server and to trigger restart if case of system crash? if so can someone explain how and where to use the feature?  if cant find any watchdog files realted to this after in update in my nanokvm. I've seen there is a update on support section related to this but im not sure how to configure it.

patschi, 2025-03-13:
Looks like this is the watchdog function, which initiates a `reboot` if "kvm_system gets stuck" for longer:
https://github.com/sipeed/NanoKVM/blob/99d1fc0a64969c98f14f83015ecc84a56718e2a1/support/sg2002/kvm_system/main/src/main.cpp#L233-L251

prasaanth-slx, 2025-03-17:
this script is not a part of default os right? can i  know how to enable watchdog service and set custom watch dog settings.  
is the `/etc/watchdog.conf ` the only file to tweak? what are parameter to be set. i want to enable the watchdog service to reset the kvm if in case it hangs which happens a lot of time when using tailscale and nrp.

closed 2025-03-17 by prasaanth-slx