sipeed/NanoKVM #379: Anyway to change mDNS hostname?

scyto, 2025-03-13:
I see how /boot/hostname.prefix sets the prefix to the autoname generation.

however i want the host name to be nano-kvm1.local and for this shotname to register with the DHCP server

how can i hard set the hostname?
is there a file i can add to /boot to do this (/etc/hostname gets reset on each boot)

pinklifeart, 2025-03-23:
> I see how /boot/hostname.prefix sets the prefix to the autoname generation.
>
> however i want the host name to be nano-kvm1.local and for this shotname to register with the DHCP server
>
> how can i hard [...]

@scyto Setting a hostname in `/boot/hostname` (doesn't exist by default) did the trick for me.

wj-xiao, 2025-05-28:
Added in version [2.2.6](https://github.com/sipeed/NanoKVM/blob/main/CHANGELOG.md#226-c83dc55-2025-04-17).