sipeed/NanoKVM #381: No Keyboard detected / Onboard Ethernet Stops working.

sidneyki, 2025-03-14:
Issue: Not Keyboard Detected / Onboard Ethernet Stops working
Model: Nano PCI KVM with POE no WIFI

Motherboard: ROG STRIX B450-F Gaming II

1st Install steps: 
install the NANO kvm in to a PCIEX1 slot on the motherboard. Only plug usb-c hub port and the other end usb-a to the computer usb port. When the power plug is plugged in to the computer the Onboard Ethernet Port starts blinking. The blue led blinks on the KVM and it boots and gets IP address. I am able to connect to the KVM and I have ran the update on it. When powering on the PC the Onboard Ethernet port can not get a link to the network. 

2nd install:
Remove the KVM from the PCIEX1 slot and put it in a blank spot. I plugged both usb cable in to the KVM. I am able to connect but when the pc boot says no keyboard detected and the Onboard Ethernet port not able to connect to the network.

When the computer boots says no keyboard detected.

the KVM Some motherboard BIOS require HID keyboard and mouse devices to have a BIOS identifier. The NanoKVM can create a BIOS under /boot to enable this feature by executing touch /boot/BIOS && restart I an error message "/bin/bash: restart: not found"

I tried:
1.7. HID Keyboard and Mouse Not Working[#](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/faq.html#HID-Keyboard-and-Mouse-Not-Working)
Use the "Reset HID" function in the web interface.
Check if the USB port is securely connected; you can check if the HID icon on the OLED is lit, or use cat /sys/class/udc/4340000.usb/state in the web terminal. If it shows not connected, it indicates a poor connection of the USB cable. Please replace the USB cable and try again.
I ran the Rest HID and it did not do anything.

I really could use some help resolving these issues.

Z2Z-GuGu, 2025-03-26:
非常抱歉给你带来了不佳的体验，我们刚才推送了最新的2.2.5版本更新，它包含了改善HID兼容性的部分，可以在更新后通过鼠标图标中的“HID Only”模式打开。在这种模式下，NanoKVM的USB会仅模拟出键盘鼠标触摸板设备，设备的兼容性会提高。
另外，新版本中“PC启动时IP消失”的问题是否解决了呢

sidneyki, 2025-03-26:
你好，我已将 Nano KVM 更新到 2.2.5 版本。我已经重启了 KVM，并重置了 HID。我还重启了电脑，并尝试使用内置键盘。我已启用 HID 仅模式。但键盘和鼠标仍然无法使用。我仍然需要帮助来解决键盘问题。此次更新并未修复问题。

Z2Z-GuGu, 2025-04-01:
Is the issue with your IP disappearing still occurring? Also, could it be caused by a wiring error? If possible, please take a photo. Additionally, I thought of another possibility: an issue with the host's drivers. In this case, restarting the HID will not work. You will need to manually uninstall all NanoKVM USB devices in the device manager and then unplug and replug the NanoKVM.