sipeed/NanoKVM #383: USB HID stops working after 10-60s (mouse + keyboard)

dokterdok, 2025-03-15:
dmesg shows the same error repeatedly, right after booting the machine, sometimes 10 seconds after, sometimes up to a minute later:

```
[  102.451818] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
[  102.451829] configfs-gadget gadget: usb_gadget_wakeup
[  102.451841] configfs-gadget gadget: usb_ep_queue error on int endpoint -11
[  102.451886] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
[  102.451918] configfs-gadget gadget: usb_gadget_wakeup
[  102.451933] configfs-gadget gadget: usb_ep_queue error on int endpoint -11
```
I've only tried accessing the BIOS for now (MS-01).
The only workaround I found is to reboot the host, or to swap the usb port. Both of these options are only brief, HID as fails again shortly after. Resetting HID from the web app doesn't work.

Using the latest v1.4.0 / 2.2.2 updates.

supczinskib, 2025-03-17:
I have exactly the same issue.
The cursor stops responding. Reloading the page helps, but after a short while the mouse cursor stops responding again.

dokterdok, 2025-03-21:
Issue persists with v 2.2.4

supczinskib, 2025-03-22:
Confirmed. There are still serious issues with the mouse functionality. Now, in absolute mode, it doesn’t work at all. Restarting either the computer or the NanoKVM doesn’t help. I have two computers and two NanoKVMs—one Lite and the other Full. In both cases, the symptom is the same. However, the mouse works in Relative mode. There are no issues with the virtual keyboard. I tested on macOS in both Safari and Chrome. I’m surprised by the silence; why do so few people mention this?

https://github.com/user-attachments/assets/1aafc80e-db33-44a6-822e-4def5ac42e89

supczinskib, 2025-03-24:
I downgraded to version 2.2.2. Then, in Device Manager, I removed all HID interface devices and all HID-compliant mice. After that, I performed a scan for hardware changes. Windows re-detected all devices, but this time it installed two HID-compliant mice instead of one, and everything started working correctly.

Z2Z-GuGu, 2025-03-26:
Hello, we just pushed the update for version 2.2.5, which includes improvements for HID compatibility. After the update, you can enable 'HID Only' mode in the mouse icon. In this mode, NanoKVM's USB will only emulate three HID devices: keyboard, mouse, and touchpad. This mode has better system compatibility. We tested it on several hosts, and using regular mode may cause HID to fail after a short time, but using 'HID Only' can provide stable operation. 
We hope to receive everyone's testing results after the update. Thank you!

supczinskib, 2025-03-26:
> Hello, we just pushed the update for version 2.2.5, which includes improvements for HID compatibility.
Thank you for the information. I installed version 2.2.5 and, regardless of the HID settings, codec, fps, etc., the video stream eventually stops. I checked—it’s not just the cursor but the entire stream. I played a video on YouTube and noticed it periodically freezes. I’m reverting back to version 2.2.2 because that one works most stably for me.

https://github.com/user-attachments/assets/0e8d827a-a9f0-4a30-86db-5bef1c35d02d

supczinskib, 2025-03-26:
And one more note. After restarting the Lite version, the power button icon in the top navigation bar remains active just like in the Full version, even though the software knows that this is not the version with that feature.

https://github.com/user-attachments/assets/beef5780-9164-4156-a928-708bcbbd0b61

Z2Z-GuGu, 2025-04-01:
Hello, after watching your video, I suspect that the frame difference detection did not detect the small changes in the mouse position. Could you try disabling the frame difference detection settings and test again?

Version 2.2.2 actually lacks support for frame difference detection, whereas this feature was restored in 2.2.5. So I suspect that the lack of detection of significant changes in the image is causing NanoKVM to stop actively sending data.

dokterdok, 2025-04-01:
Issue persists with 2.25. Tried the HID-only mode. Same issue. My test case is still the same:
Using the Minisforum MS-01, power up, press DEL to enter BIOS, mouse and keyboard become unresponsive after a few seconds.

supczinskib, 2025-04-02:
> Hello, after watching your video, I suspect that the frame difference detection did not detect the small changes in the mouse position. Could you try disabling the frame difference detection settings and test again?

You were right. Disabling that feature solved the problem.

https://github.com/user-attachments/assets/249c633f-0fcd-4009-b3ff-0503f9011fb7

Tom60chat, 2025-04-21:
Didn't for me :/