sipeed/NanoKVM #384: Very high delay in mouse movement (but not mouse clicks and also not keyboard input)

kunom, 2025-03-16:
For testing, I run NanoKVM against a Raspberry Pi 2 B.

While keyboard input works very well, using the mouse is barely feasible for me:
- In absolute mode, the remote pointer position is adjusted with roughly 2-3 Hz and eventually lags several seconds (5-15) behind the local pointer to reach a target position. But not for mouse clicks, they seem to get executed remotely right on local capture. This renders mouse input very unreliable, you constantly have to wait for the remote cursor to arrive at the intended location.
- In relative mode, I am barely able to move the mouse pointer at all.

Is this intended behavior? If not, please tell me how to provide more debugging input.

The NanoKVM device runs locally on the same switch (<1ms ping time), but only with a 10 MBit link.

kunom, 2025-03-17:
When looking at the Network panel, the websocket messages (mouse move) seem to be sent to the backend in rapid sequence.

UWH-David, 2025-03-18:
I am seeing a similar situation and I am connected over the NIC as well. This is frustrating and not something I'd want to do on a daily basis.

sirmeili, 2025-05-09:
Anyone find a solution for this? The lag makes this completely unusable

kunom, 2025-05-09:
There are hundreds of entries in _/var/log/messages_ which seem to be produced during mouse movement:
```
May  9 16:09:08 kvm-1b9b kern.info kernel: [  683.332874] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
May  9 16:09:08 kvm-1b9b kern.err kernel: [  683.332893] configfs-gadget gadget: usb_gadget_wakeup
May  9 16:09:08 kvm-1b9b kern.info kernel: [  683.349392] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
May  9 16:09:08 kvm-1b9b kern.err kernel: [  683.349408] configfs-gadget gadget: usb_gadget_wakeup
May  9 16:09:08 kvm-1b9b kern.info kernel: [  683.404534] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
May  9 16:09:08 kvm-1b9b kern.err kernel: [  683.404552] configfs-gadget gadget: usb_gadget_wakeup
May  9 16:09:09 kvm-1b9b kern.info kernel: [  683.717621] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
May  9 16:09:09 kvm-1b9b kern.err kernel: [  683.717639] configfs-gadget gadget: usb_gadget_wakeup
May  9 16:09:09 kvm-1b9b kern.info kernel: [  684.456714] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
```

sirmeili, 2025-05-12:
I happen to own a JetKVM and use remote desktop/VNC/etc. quite regularly. I've never had this issue until I've started using the NanoKVM. There really is no reason this should be a thing. I've also noticed that the keyboard has a lag as well.

I hope a fix is coming out soon. I can help provide logs if that is needed. Just let me know.

mcr-ksh, 2025-05-26:
This sounds all nice but the fact that you have to tweak the host and not the KVM is bothering. This should serve as a workaround only until the KVM is fixed.