sipeed/NanoKVM #385: PC wont boot with with NanoKVM connected

r0gerthat7, 2025-03-16:
Have a Lenovo PC that wont boot if NanoKVM PCIe POE is connected vis USB when first plugged in.

Steps to reproduce:

1. PC powered off and unplugged
2. NanoKVM installed, connected via USB and powered up via POE
3. Plug PC power cord in
4. PC beeps following POST test
5. Turn on PC via switch
6. PC fails to load to BIOS with error codes 003E / BIOS B253

To resolve:
1. PC powered off and unplugged
2. NanoKVM installed and powered up via POE
3. Plug PC power cord in
4. PC beeps following POST test
5. NOW connect NanoKVM in via USB
6. Turn on PC via switch
7. PC boots normally

- Behavior is the same for all USB ports (Internal 2.0 header, external 2.0 / 3.0)
- I have already disabled Virtual Disk and Virtual Network in Nanokvm settings
- Problem does not happen if you power PC on/off without removing main power (ie reboot / power off + on)

There is something the PC doesnt like about the nanokvm if its plugged in before power is plugged in / POST.

chiconws, 2025-03-17:
I have the same issue: my PC won’t boot past the BIOS splash screen when the NanoKVM is connected via USB (whether internally or externally). If I boot the PC first and then connect the NanoKVM via USB, it works. However, if the machine reboots for any reason, it gets stuck on the BIOS splash screen. Since the PC is in a remote location (necessitating the use of a remote KVM), I can’t fully utilize the NanoKVM (PCI Edition with PoE).

chiconws, 2025-03-25:
@r0gerthat7 did you upgraded to  2.2.4? A few days ago they committed [`ee77f7b`](https://github.com/sipeed/NanoKVM/commit/ee77f7bed1577ed529d8d33d374e7be70c3d5add) (Fix USB initialization issue), but I can't test it right now.

chiconws, 2025-03-26:
I'm only going to be able to test it next month, but I'll keep you posted. Thank you in advance.

r0gerthat7, 2025-03-26:
@chiconws The issue for me was NOT fixed with 2.2.4

@Z2Z-GuGu I updated to 2.2.5 and enabled HID-only Mode. No improvement.

Z2Z-GuGu, 2025-04-01:
"To enable the host to boot first by temporarily disabling the USB interface and then re-enabling USB functionality, please test with the following commands:

```shell
# Disable USB functionality
echo > /sys/kernel/config/usb_gadget/g0/UDC
# Please manually power on the system
# Re-enable USB
ls /sys/class/udc/ | cat > /sys/kernel/config/usb_gadget/g0/UDC
```

chiconws, 2025-04-05:
Don't know if it was the 2.2.4 or 2.2.5, but my PC won't get stuck on the BIOS splash screen anymore. I turned on the HID Only option. Thank you.