sipeed/NanoKVM #387: nanokvm remote os update

prasaanth-slx, 2025-03-17:
is there a way to do remote os update if the nanokvm product is used for production? if not will can we expect future upgrades on this?

KWottrich, 2025-03-18:
Given that application updates are already possible remotely, I assume you're asking about the ability to update the **system image** remotely, correct? (I'm not a dev, so I can't answer your question, just wanted to clarify.)

prasaanth-slx, 2025-03-19:
yes im asking about the system image

ucki007, 2025-03-19:
Hi, 

I think there still is no official way to do that, but somebody found out how to do that: #219 

I haven't verified, that this is working, but some reports say it is.

KWottrich, 2025-03-19:
> Hi,
> 
> I think there still is no official way to do that, but somebody found out how to do that: [#219](https://github.com/sipeed/NanoKVM/issues/219)
> 
> I haven't verified, that this is working, but some reports say it is.

That's a pretty dangerous way to do it. The steps outlined are:

1. Unpack the boot and rootfs filesystems from the firmware image on a second computer
2. Remotely delete the `/boot` directory from the NanoKVM (!!)
3. Copy the new boot directory into `/boot` on the NanoKVM
4. Copy (and overwrite) the rootfs into `/` on the NanoKVM

If anything goes wrong after step 2, you can't use your NanoKVM, and it may be completely unbootable if anything in step 3 fails.  Since this whole discussion is about how to do this for a production (and not local) KVM, it should probably something that's resilient to errors.  That link is a great starting point, and certainly works if you're willing to risk it, but it would be much better to do something local to the KVM itself like an A/B partition swap or something.

CRCinAU, 2025-03-21:
Agreed - it isn't a perfect method, but it works.

The official way is to remove the SD Card, flash the image to it and reinstall it back into the NanoKVM - which would still be the fallback if you managed to mess up the remote update.

ucki007, 2025-03-21:
> Agreed - it isn't a perfect method, but it works.
> 
> The official way is to remove the SD Card, flash the image to it and reinstall it back into the NanoKVM - which would still be the fallback if you managed to mess up the remote update.

I learned, you do not even have to remove the SD card.

You can switch the NanoKVM to "USB stick mode" by pressing and holding the button next to the USB port during power on:
[image]

 You can even so that on the NanoKVM Cube with the included SIM card removal tool. 
See image: stick it in the hole (hehehehe), pull up and you will feel a slight "click".

[image]

rbm78bln, 2025-04-21:
Okay, "dangerous" is more like a way of saying... of course I wouldn't do this with a wonky internet connection, but:

I guess with a little bit of fiddling around one could also easily do this:
> sync && echo /dev/mmcblk0p* | xargs -n1 mount -o remount,ro && echo u > /proc/sysrq-trigger && \\
> curl 'https://github.com/path/to/someimage.img.xz ' | unxz -fc | dd bs=1M iflag=fullblock oflag=direct of=/dev/mmcblk0 && \\
> sync && echo b > /proc/sysrq-trigger

But yes, of course, a ready-to-use script by the developers and a button on the UI would be fancy. 😎