sipeed/NanoKVM #390: Image shaking, floating

netandreus, 2025-03-17:
**Problems:**

I just installed NanoKVM Lite and faced with these issues:
- the image is shaking, floating
- the image is cropped on the left, so that the Start button is not available on Windows.

**Screenshots:**

[image]

**Environment:**

[image]

Zepan, 2025-03-25:
Hi, we will soon update 2.2.5, you can update then and check if it fixed

wj-xiao, 2025-03-25:
At resolutions of `800x600` and `640x480`, the display may not working correctly. 
It should works fine with the `Automatic`, `1920x1080`, and `1280x720` resolutions. Could you please try these three settings?

We are working to resolve this issue and expect to include the fix in next month's release.

UWH-David, 2025-03-26:
> At resolutions of `800x600` and `640x480`, the display may not working correctly. It should work fine with the `Automatic`, `1920x1080`, and `1280x720` resolutions. Could you please try these three settings?
> 
> We are working to resolve this issue and expect to include the fix in next month's release.

Many including myself are seeing very large latency even when using a network cable. Any fix for that?

netandreus, 2025-03-26:
I try all resolutions, results a re the same.
Also trying to update NanoKVM.

```
Image Version
v1.0.0
Application Version
2.2.5
```

Same result :(

LeonStraathof, 2025-04-18:
> I tried all resolutions, results a re the same. Also trying to update NanoKVM.
> 
> ```
> Image Version
> v1.0.0
> Application Version
> 2.2.5
> ```
> 
> Same result :(

Hi Netandreus, can you provide more details: like are you using a normal passive short HDMI cable or are you using splitters extenders or any other device in between? What is the brand/model of the computer/videocard you try to capture?

TreeFrogTinkerer, 2025-05-27:
I am experiencing he same issue.  I use it to build machines en mass so I've used it with a good 50+ individual computers - vast majority are Intel/ASUS NUC models.  

Some more details. Always using a passive HDMI cable.  It happens intermittently on NUC12, NUC10, NUC14 -- they all are using onboard Intel graphics from their respective CPU generation.  If I reboot the NanoKVM from the menu it persists and stays wonky.  If I reboot the NanoKVM by pulling power it usually comes up correctly after the first time thought sometimes it takes 2 hard reboots.  

If I reboot the Windows machine that was working correctly with NanoKVM it may or may not come back up working correctly.  A random amount of times it will be wonky like above.  It always works in BIOS even when Windows is wonky.

Hopefully that adds some details for those trying to fix it.