sipeed/NanoKVM #391: Problem with video image

Mendex777, 2025-03-17:
The image broke after starting the installation and it happens every time.
[image]

[image]

Mendex777, 2025-03-17:
[image]

netandreus, 2025-03-18:
I have the same, man :(
https://github.com/sipeed/NanoKVM/issues/390

AT-StephenDetomasi, 2025-03-18:
Can you use the Terminal UI in the meantime?

Mendex777, 2025-03-18:
If you are talking about the item "install Proxmox VE (Terminal UI)" then the situation is exactly the same as in the photo above.

Mendex777, 2025-03-18:
I also tried to boot from other bootloaders that I have. For example, with Kubuntu, there are no problems, everything is displayed correctly.
But with loading AdminPE, there are similar problems with video.
From this I conclude that the problem is not specifically in the installation Proxmox.

Z2Z-GuGu, 2025-03-21:
The Proxmox VE installation software may have output an unsupported resolution, causing screen tearing when NanoKVM reads these images. I am trying to modify the HDMI EDID to resolve this issue.

Mendex777, 2025-03-21:
The problem is not only with Proxmox VE, there are other images that cannot boot properly. For example, AdminPE.