sipeed/NanoKVM #393: IPMI Support

SagnikSRHUSE, 2025-03-18:
Is it possible to use ipmitool to manage the NanoKVM?

brainkiller, 2025-03-18:
+1 would be useful for automation

mark99i, 2025-03-23:
+1 would be useful for automation

harp077, 2025-10-04:
+1 would be useful for automation !!!

brainkiller, 2026-02-10:
Foregive me for my ignorance.
I'm running 2.3.4 and I cannot find the location /etc/ipmi in my installation.
Should I downgrade to v1.4.2 to make this work or are there plans to re-implement this ?

matusvlcek, 2026-09-13:
> Is there anyone who can clarify this ?

I was wondering the same thing. I was running App version 2.4.3, but the Image version (yes, those are two different things) was 1.4.1. I had to flash it manually: https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html.