sipeed/NanoKVM #394: Lenovo M710q Tiny Power control

akmadov, 2025-03-18:
Hi! I have bought the full version of the NanoKVM and would like to be able to power my Lenovo M710q tiny with the Alt+P keyboard combination that is supported by the PC. However, It does not work when trying to do it through the KVM, neither does using the onscreen keyboard in the web interface, nor does using the physical keyboard that is connected to my PC (not the remote that I am viewing through the KVM). What would be your advice? Thank you!

ucki007, 2025-03-19:
I did not find the information in any documentation. I simply disassembled the PC and followed the lanes on the motherboard and then validated with a multimeter that using the original power button shortened the the identified contact points.

akmadov, 2025-03-27:
I'm back to report two things:
1. No such contacts are available on the board of my M710Q tiny, unfortunately. The only this I can do is solder directly to the power button leads, which is doable, but requires a bit of precision to not accidentaly jump the wrong contacts.
2. The HID-only mode introduced with version 2.2.5 of the NanoKVM has fixed the Alt + P combo not working. Now I can turn on the PC remotely with Alt + P. The only thing left is to figure out the force shutdown by holding down the power button (on the PC itself). But I may resort to using a smart plug that will just cut the power to the PC if the need presents itself, thus avoiding any soldering and potential for screw ups.