sipeed/NanoKVM #395: Video image very poor and streaky, nanoKVM therefore completely unusable

derhalbe, 2025-03-18:
Video image very poor and streaky, therefore unusable - see picture attached,
flashing firmware v1.4.0 with 2.2.0 and update to 2.2.2 didn't solve the issue,
also testing different resolutions, qualities, FPS, HDMI cables etc. didn't help
PLEASE HELP
thank you

[image]

netandreus, 2025-03-19:
I have the same #390

derhalbe, 2025-03-19:
> I have the same [#390](https://github.com/sipeed/NanoKVM/issues/390)

there is one difference,
in my case the streaks are stable, not flury and dynamically

netandreus, 2025-03-19:
Hmm.... still waiting for some response from developers...

derhalbe, 2025-03-25:
> Hi, what resolution are you using?

"also tested different resolutions, qualities, FPS", screenshots were done with 1080p

derhalbe, 2025-04-04:
aparently, my seller exchanged it

closed 2025-04-04 by derhalbe