sipeed/NanoKVM #396: Poor Video and Latency

UWH-David, 2025-03-18:
Grabbed a NanoKVM Lite. The documentation could use a ton of work. For those wondering, you want to grab the latest .img.xz file from there, ignore all the nanaokvm tarballs as the interface can update to those when running under the settings cog, "Check for Updates". 

What I am noticing, even when connected directly over 100mbs, is the video quality is pretty low. White are grey and the detail is certainly lost. I've tried changing video mode between MJPEG, H.264, lowering resolution, setting quality to lossless, When it comes to the mouse and keyboard, the latency is pretty bad too. This might be great for emergency issues fixing a box remotely but not something I'd want to use daily. Has anyone had any success on improving this situation?

I did order a JetKVM through Kickstarter and I am anxiously awaiting that product which looks promising.

UWH-David, 2025-03-26:
Any updates to address this issue?

binboum, 2025-03-30:
I have the same problem after 2.2.1, can you try to restore to this version?

UWH-David, 2025-03-30:
Sure, what's the downgrade process on these? 

I did try:

```
/usr/src # wget http://github.com/sipeed/NanoKVM/releases/download/2.2.1/nanokvm_2.2.1.tar.gz
Connecting to github.com (140.82.116.3:80)
wget: not an http or ftp url: https://github.com/sipeed/NanoKVM/releases/download/2.2.1/nanokvm_2.2.1.tar.gz
```